RT-Thread/rt-thread #7005: 这

BeDay, 2023-03-02:


Guozhanxin, 2023-03-24:
有好的解决方法嘛？

BeDay, 2023-03-24:
> 有好的解决方法嘛？

1.ch是 “\0” 跳过不存入 recv_buffer 
2.判断结束标识时 判断 recv_buffer 长度 并清空

 `
        result = server->get_char(server, &ch, RT_WAITING_FOREVER);
        // 判断字符读取是否成功
        if (result != RT_EOK){
            at_server_printfln("");
            at_server_printfln("get char error %d", result);
            continue;
        }

      if (ESC_KEY == ch)
        {
            break;
        }

        if ( RT_NULL == ch)
        {
            // 空字符 \0 不接收
            continue;
        }
      ...........
     
     // 判断不是 end_mark 同时判断 recv_buffer 满了 跳转清空
       if(!strstr(server->recv_buffer, server->end_mark))
        {
            if(server->cur_recv_len >= AT_SERVER_RECV_BUFF_LEN){
                at_server_printfln("");
                at_server_printfln("buff full eq %d clear buffer", AT_SERVER_RECV_BUFF_LEN);
                goto __retry;
            }
            continue;
        }

`

mysterywolf, 2023-05-14:
此PR已经解决第一个问题，第二个问题可否提交一个PR？谢谢！

armink, 2023-05-14:
既然是 AT 这种字符协议，就不应该会传入 `\0` ，楼主可否说下遇到问题的场景

wenxingpt, 2023-05-15:
> 此PR #7370 已经解决第一个问题，第二个问题可否提交一个PR？谢谢！

If we ensure that there is no '\0' input, then no further changes will be necessary, maybe we can consider this thread completed~

closed 2023-05-15 by mysterywolf