RT-Thread/rt-thread #7012: 在RT-Thread中使用Atomic

BernardXiong, 2023-03-03:
当需要高性能时，atomic操作，无锁操作是至关重要的。在RT-Thread中引入atomic可以是这样：

* 引入几个配置条件宏
  * RT_USING_HW_ATOMIC - 使用hardware层实现的atomic
  * RT_USING_STD_ATOMIC - 使用stdatomic.h中实现的atomic
  * 以上两者都不定义，则采用开关中断方式的`软`实现

`rtdef.h`中

```c
#if defined(RT_USING_STD_ATOMIC)
#include <stdatomic.h>
typedef atomic_size_t rt_atomic_t;
#else
typedef volatile rt_base_t rt_atomic_t;
#endif
```

`rtthread.h`中

```c
#if defined(RT_USING_HW_ATOMIC)
#define rt_atomic_add(ptr, v) rt_hw_atomic_add(ptr, v)
/* 宏定义到libcpu下的hw_atomic实现 */
#elif defined(RT_USING_STD_ATOMIC)
#define rt_atomic_add(ptr, v) atomic_fetch_add(ptr, v)
/* 宏定义到c11中的atomic实现 */
#else
#endif
```

具体实现，开关中断方式`软`实现 `atomic.c` in `rt-thread/src`

否则，
* RT_USING_HW_ATOMIC => 使用cortex-m/a/riscv下的hw_atomic.c实现
* RT_USING_STD_ATOMIC => 使用libc中的C11 atomic实现；

-----
参考PR
* #6931 
* #6197

mysterywolf, 2023-03-03:
赞成赞成

BernardXiong, 2023-03-04:
c11的atomic详细说明：

https://github.com/zenny-chen/C11-atomic-operations-in-detail

对于RT-Thread来说，atomic类型会取`atomic_size_t`，即和处理器位宽相关的类型（标准类型对应到`size_t`）。

```c
#include <stdatomic.h>
typedef atomic_size_t                   rt_atomic_t;

#define rt_atomic_add(ptr, v)           atomic_fetch_add(ptr, v)
#define rt_atomic_sub(ptr, v)           atomic_fetch_sub(ptr, v)
#define rt_atomic_or(ptr, v)            atomic_fetch_or(ptr, v)
#define rt_atomic_xor(ptr, v)           atomic_fetch_xor(ptr, v)
#define rt_atomic_and(ptr, v)           atomic_fetch_and(ptr, v)
#define rt_atomic_cas(ptr, v, d)        atomic_compare_exchange_strong(ptr, v, d)

#define rt_atomic_set(ptr, v)           atomic_store(ptr, v)
#define rt_atomic_get(ptr)              atomic_load(ptr)
```

对于size_t来说，后续也需要在rt_kprintf中加入`%zd`的支持。

closed 2023-03-21 by mysterywolf