RT-Thread/rt-thread #7019: Question about the use of extern in nuvoton bsp header file

cy023, 2023-03-05:
Hi there,

I have a question about the use of `extern` in the bsp/nuvoton/libraries/m480/StdDriver/inc/nu_fmc.h header file. Specifically, I noticed that the `extern` keyword is used in function declarations. However, as far as I know, C already assumes extern by default when declaring functions.

I would like to understand why `extern` is used in these function declarations, and whether it is necessary.

Thanks in advance for any clarification you can provide!

BernardXiong, 2023-03-05:
Hello, 

Yes, `extern` has no meaning when declaring functions in the header file.

wosayttn, 2023-03-06:
Hello,

Yes, no other meaning.
Just gives a reminding that is a public function. Orz.

cy023, 2023-03-06:
Thank you very much for your help and guidance!

closed 2023-03-09 by cy023