RT-Thread/rt-thread #7020: RT-Thread Task List | RT-Thread 任务清单

mysterywolf, 2023-03-05:


BernardXiong, 2023-03-05:
* Task: atomic feature for RT-Thread #7012

mysterywolf, 2023-03-05:
- [ ] 软件包索引choice块内禁止使用if块，否则Kconfig C前端无法识别 https://github.com/RT-Thread/packages/issues/1564#issuecomment-1455218946

supperthomas, 2023-03-08:
- [ ] DOXYGEN warning修复 #6857

BernardXiong, 2023-03-13:
* [ ] 对调度器进行拆分，分成公共部分，单核部分，SMP多核部分；

这样在调度器层面，如果仅针对单核大多数场合只看到单核的代码；针对多核时，才会看到更多多核的代码。

可以考虑把schedule.c拆分成三个：
* scheduler_common.c，up/mp下公共的文件实现；
* scheduler_up.c
* scheduler_mp.c

BernardXiong, 2023-03-13:
* [ ] 消息邮箱消息队列返回消息长度的问题 https://github.com/RT-Thread/rt-thread/issues/5730

BernardXiong, 2023-03-13:
* [ ] 加入tls（Thread Local Storage）特性，可以在tls基础上加入更多的辅助功能

mysterywolf, 2023-03-16:
```
/* global errno in RT-Thread */
static volatile int __rt_errno;

rt_err_t rt_get_errno(void)
{
    rt_thread_t tid;

    if (rt_interrupt_get_nest() != 0)
    {
        /* it's in interrupt context */
        return __rt_errno;
    }

    tid = rt_thread_self();
    if (tid == RT_NULL)
        return __rt_errno;

    return tid->error;
}

void rt_set_errno(rt_err_t error)
{
    rt_thread_t tid;

    if (rt_interrupt_get_nest() != 0)
    {
        /* it's in interrupt context */
        __rt_errno = error;

        return;
    }

    tid = rt_thread_self();
    if (tid == RT_NULL)
    {
        __rt_errno = error;

        return;
    }

    tid->error = error;
}

__rt_errno 的原始数据类型是 int. 

typedef long                            rt_base_t;      /**< Nbit CPU related date type */
typedef rt_base_t                       rt_err_t; 

rt_err_t 的类型是 long 

所以这里就是 int 和 long 直接就转了，在 32 位平台int 和 long 都是 32 位，在 64 位的时候，long 就是 64 位了
```

BernardXiong, 2023-03-18:
可以考虑重点维护一些BSP：
* MCU
* WiFi类芯片
* MPU
  * ARM Cortex-A
  * ARM Cortex-A64
* RISCV相关典型芯片
  * RV64 with smart

BernardXiong, 2023-04-02:
update list for v5.0.1

* DFS v2.0 for rt-smart
  * #7156 Kconfig for DFS v2.0
* Optimize gdb debugging behavior
  * #7033

mysterywolf, 2023-04-29:
- [ ] 补充如下文件夹及其子文件夹下.c文件函数头注释：
 - https://github.com/RT-Thread/rt-thread/tree/master/components/libc

dqjay110, 2023-05-08:
考虑增加DMA驱动框架

mysterywolf, 2023-06-26:
> 考虑增加DMA驱动框架

考虑交给底层驱动自由实现？而不是提供一套统一的DMA抽象。

BernardXiong, 2023-08-05:
> * [ ]  加入tls（Thread Local Storage）特性，可以在tls基础上加入更多的辅助功能

这部分可以尝试先添加一些测试用例，看看这部分的tls如何使用，影响包含哪些。

BernardXiong, 2023-08-05:
* [ ] 加入IP化的设备驱动框架，包括设备树，pinctrl，clk等基础框架；
  * [ ] pinctrl框架
  * [ ] clk框架
* [ ] tty更新和完善

enkiller, 2023-08-30:
TLS 特性目前在推进吗？大概是什么情况？

BernardXiong, 2023-09-03:
> TLS 特性目前在推进吗？大概是什么情况？

应该暂时没有。tls特性在smart上遇到了更多，不过目前看起来，似乎都还ok。