RT-Thread/rt-thread #7038: fal处理片内flash擦除了其它地址

zmoth, 2023-03-10:
芯片: stm32F405RG
RT-Thread版本: 4.0.5
fal版本 v0.5.0

```c
int write_1(const uint8_t *buf, size_t size)
{
    fal_partition_erase(partition, 0, size);
    return fal_partition_write(partition, 0, buf, size);
}

int write_2(const uint8_t *buf, size_t size)
{
    fal_partition_erase(partition, 40, size);
    return fal_partition_write(partition, 40, buf, size);
}
```
在调用`write_2()`后，1中的数据也被擦除了(被置为0xFF). 输入的buf并没有越界。

目前问题应该是由`fal_partition_erase()`造成的，如果我注释它，那么该存储区间只能修改一次。

> 虽然比较老，但是我看fal_partition_erase()这部分没有做过修改，并且没人提出这个问题

muaxiaohei, 2023-03-23:
你的片内flash擦除一页应该大于40个字节吧？你擦除40开始的内存地址，其实也会把前面的内容擦除，我记得fal应该不存在缓冲区，来先提取前面的内容后，再擦除的，并将内容写回的功能

muaxiaohei, 2023-03-23:
这种功能一般都是应用层实现的，如果底层实现了，会导致不必要的内容占用

Guozhanxin, 2023-03-24:
> 这种功能一般都是应用层实现的，如果底层实现了，会导致不必要的内容占用

是的，这种功能应该应用层来做。

closed 2023-03-24 by Guozhanxin

zmoth, 2023-03-24:
谢谢！