RT-Thread/rt-thread #7041: pin设备驱动部分引脚编号的处理方式与现有模拟i2c使用pin编号衔接不当

charlown, 2023-03-12:
问题描述，以stm32为例
在drv_gpio.c中
#if defined(SOC_SERIES_STM32MP1)
#if defined(GPIOZ)
#define gpioz_port_base (175) /* PIN_STPORT_MAX * 16 - 16 */
#define PIN_STPORT(pin) ((pin > gpioz_port_base) ? ((GPIO_TypeDef *)(GPIOZ_BASE )) : ((GPIO_TypeDef *)(GPIOA_BASE + (0x1000u * PIN_PORT(pin)))))
#else
#define PIN_STPORT(pin) ((GPIO_TypeDef *)(GPIOA_BASE + (0x1000u * PIN_PORT(pin))))
#endif /* GPIOZ */
#else
#define PIN_STPORT(pin) ((GPIO_TypeDef *)(GPIOA_BASE + (0x400u * PIN_PORT(pin))))
#endif /* SOC_SERIES_STM32MP1 */

在board/Kconfig中
menuconfig BSP_USING_I2C
        bool "Enable I2C BUS"
        default n
        select RT_USING_I2C
        select RT_USING_I2C_BITOPS
        select RT_USING_PIN

        if BSP_USING_I2C
            config BSP_USING_I2C1
                bool "Enable I2C1 Bus (User I2C)"
                default n
                if BSP_USING_I2C1
                    comment "Notice: PB9 --> 25; PB8 --> 24"
                    config BSP_I2C1_SCL_PIN
                        int "i2c1 SCL pin number"
                        range 1 176
                        default 24
                    config BSP_I2C1_SDA_PIN
                        int "i2c1 SDA pin number"
                        range 1 176
                        default 25
                endif
        endif

以前数组存放的方式可以查看对应的编号，但现在就没法知道

mysterywolf, 2023-03-12:
.你可以通过rt kprintf GET_PIN的方式打印出引脚编号来获取

charlown, 2023-03-13:
i2c配置采用管脚名, 用rt_pin_get，ops.data不传at32_soft_i2c_config而是传at32_i2c，不过改动多些

[image]

```c
struct at32_soft_i2c_config
{
    const char *scl_name; //"PA.09"
    const char *sda_name; //"PA.10"
    const char *bus_name;
};

struct at32_i2c
{
    rt_base_t scl;
    rt_base_t sda;
    struct rt_i2c_bit_ops ops;
    struct rt_i2c_bus_device i2c_bus;
};

for (int i = 0; i < obj_num; i++)
    {

        i2c_obj[i].scl = rt_pin_get(soft_i2c_config[i].scl_name);
        i2c_obj[i].sda = rt_pin_get(soft_i2c_config[i].sda_name);

        i2c_obj[i].ops = at32_bit_ops_default;
        i2c_obj[i].ops.data = (void*)&i2c_obj[i];
        i2c_obj[i].i2c_bus.priv = &i2c_obj[i].ops;

        

        at32_i2c_gpio_init(&i2c_obj[i]);
        result = rt_i2c_bit_add_bus(&i2c_obj[i].i2c_bus, soft_i2c_config[i].bus_name);
        RT_ASSERT(result == RT_EOK);
        at32_i2c_bus_unlock(&i2c_obj[i]);

        LOG_D("software simulation %s init done, pin scl: %d, pin sda %d",
        soft_i2c_config[i].bus_name,
        soft_i2c_config[i].scl,
        soft_i2c_config[i].sda);
    }
```

charlown, 2023-03-13:
看看这种解决办法如何，pin设备那个引脚编号，我觉得"PF.09"比“PF.9”好些

mysterywolf, 2023-03-13:
之前有提到这种方案，但是之所以一直采用数字编号的原因是因为 用户在给引脚名称的时候可能会打出各种稀奇古怪的格式  PA9 PA09 PA.9 PA.09 还有些BSP可能不是PA09这种命名方式，一旦交给厂商自己去搞命名很有可能最后就乱掉了，所以还是保持纯数字的方式。
用纯数字的方式，虽然每个厂商的计算方式不一样，但是只要通过rt_kprintf(GET_PIN())获取到的数字肯定是对应的，用户只需要复制这个数字进去就可以了，可以不需要知道数字的内在逻辑或者含义。

closed 2023-03-14 by mysterywolf