RT-Thread/rt-thread #7152: [stm32f429-st-disco] compile error

kamejoko80, 2023-04-01:
Hello, 

I've tried to build bsp of stm32f429-st-disco and got the below error message:

`
stm32f429-st-disco % scons
scons: Reading SConscript files ...
TypeError: can only concatenate deque (not "list") to deque:
  File "/Users/127030/Workspace/RT-Thread/rt-thread/bsp/stm32/stm32f429-st-disco/SConstruct", line 60:
    DoBuilding(TARGET, objs)
  File "/Users/127030/Workspace/RT-Thread/rt-thread/tools/building.py", line 808:
    local_group(group, objects)
  File "/Users/127030/Workspace/RT-Thread/rt-thread/tools/building.py", line 766:
    CPPDEFINES = Env.get('CPPDEFINES', ['']) + group.get('LOCAL_CPPDEFINES', [''])
`

Then I switched to build BSP of stm32f401-st-nucleo and I didn't see that issue, maybe there are some things wrong in the python configuration but I don't know how to fix them. Is there someone else who has the same issue as me?

supperthomas, 2023-04-01:
https://github.com/RT-Thread/rt-thread/actions/runs/4573155780/jobs/8090526922

you can check the result of ci build.  it seem ok

kamejoko80, 2023-04-01:
Thank you @supperthomas for a good reference link.
Sorry, I didn't provide enough information, my PC is MAC OS, also tried to compile with other Ubuntu PC without any issues.
I need to check the environment again.

supperthomas, 2023-04-01:
good. Welcome to share the method how to deal with this problem which we think is very helpful. Thank you

huqifj, 2023-06-14:
I had the same problem. 
I found that `type(Env.get('CPPDEFINES', ['']))` returns `deque` but `type(group.get('LOCAL_CPPDEFINES', ['']))` returns a list.
`deque` can't be `+` with `list`.
I can fix it like this: `CPPDEFINES = Env.get('CPPDEFINES', ['']) + group.get('LOCAL_CPPDEFINES', [''])`.

ufbycd, 2024-07-12:
> I had the same problem. I found that `type(Env.get('CPPDEFINES', ['']))` returns `deque` but `type(group.get('LOCAL_CPPDEFINES', ['']))` returns a list. `deque` can't be `+` with `list`. I can fix it like this: `CPPDEFINES = Env.get('CPPDEFINES', ['']) + group.get('LOCAL_CPPDEFINES', [''])`.

deque can not `+` to list, you can fix it like this: 
```python
CPPDEFINES = list(Env.get('CPPDEFINES', [''])) + group.get('LOCAL_CPPDEFINES', [''])
```

huqifj, 2024-07-22:
@ufbycd Uh, you are right, maybe I made a typo, thanks for pointing that out.