RT-Thread/rt-thread #7173: SMP 下 rt_hw_interrupt_disable 的实现问题

grissiom, 2023-04-04:
当前 SMP 下面 rt_hw_interrupt_disable 的实现是禁用局部的中断，禁用其他核的调度器。这个有个问题是其他核的中断其实是没有被禁用的，所以其实如果在 ISR 里面用了资源的话其实是保护不到的…………

不过鉴于 rt_hw_interrupt_disable 用的太多了，用一次就禁用其他核的中断也不合适，所以最好的方式或许还是：

  SMP 模式下每个 IPC 对象内嵌一个 spin lock，rt_hw_interrupt_disable 从 rt_cpus_lock 切换成 rt_spin_lock_irqsave，等在各个对象的锁里而不是全局锁。

  非 SMP 模式下退化成当前的模式

貌似没捷径可以走哇

BernardXiong, 2023-04-04:
后面也想着是把`rt_hw_interrupt_disable`都少使用，而是用spinlock_lock_irqsave细粒度锁的方式。

grissiom, 2023-04-05:
所以感觉这块应该是代码统一按照 SMP 的模式写，根据配置实际的实现退化成 UP。

后面估计多核的 MCU 也会越来越多的……

grissiom, 2023-04-05:
另外就是当前对于 SMP 的测试是在哪个平台上面进行的？改了之后通过哪些测试集测试可以确认没有改坏掉？

aozima, 2023-04-05:
SMP下写代码得忘了关中断这回事才行。。

BernardXiong, 2023-04-06:
> 另外就是当前对于 SMP 的测试是在哪个平台上面进行的？改了之后通过哪些测试集测试可以确认没有改坏掉？

可以做iperf测试，并且任务不绑定核。然后也可以做些io相关的测试