RT-Thread/rt-thread #7176: [bug]  Rt-thread file flag conflict with little fs

dingiso, 2023-04-05:
Hi RT-Thread 

您好，RT-Thread的开发者，我的同学在使用 rt-thread 时，发现 一处 littlefs 和 rt-thread的描述符不对应的问题，详细可以看一下 club 中的推文

https://club.rt-thread.org/ask/question/7a40cc639a77a644.html

我核对了一下，https://github.com/littlefs-project/littlefs/blob/6a53d76e90af33f0656333c1db09bd337fa75d23/lfs.h#L127-L130 确实与 lfs.h 的对应不一样

他提交了一个PR https://github.com/RT-Thread/rt-thread/pull/7174 

他对rt-thread不是十分了解，抱歉造成了困扰

请问 这种问题 应该怎么解决 

Thanks,
Dingisoul

mysterywolf, 2023-04-06:
谢谢报告 会考虑一下怎么处理

BernardXiong, 2023-04-07:
需要处理littlefs那边的，而不是这边的。常规的定义都是目前RTT upstream这样的模式，可以参看下linux，bsd的模式。

做这样定义调整后，确实会相对容易处理。但目前这块已经成为业界事实，所以有种做法就是对这个参数进行+1，然后变成不同的值进行相关的判断。

dingiso, 2023-04-07:
ok 那我先去 littlefs 那边提交issue 然后让我的同学 统计一下会使用到的该参数 和需要 littlefs 对接的内容，再做进一步讨论

closed 2024-12-15 by Rbb666