RT-Thread/rt-thread #7192: GD32初始化关中断问题

CornerOfSkyline, 2023-04-07:
https://github.com/RT-Thread/rt-thread/blob/master/bsp/gd32/arm/libraries/GD32F4xx_Firmware_Library/CMSIS/GD/GD32F4xx/Source/system_gd32f4xx.c#L161-L162

如上，原库文件中并无关中断操作，这个关中断操作是否会导致系统无法运行？

CornerOfSkyline, 2023-04-07:
@zihao-yuan

zihao-yuan, 2023-04-08:
我没有对GD32库文件做任何形式的修改，实测GD32F470是正常工作的，我提交时GD32的库版本为3.0.3确实是有这个关中断操作的，现在最新的3.0.4在169行也是有关中断操作的。
[image]

CornerOfSkyline, 2023-04-10:
> 我没有对GD32库文件做任何形式的修改，实测GD32F470是正常工作的，我提交时GD32的库版本为3.0.3确实是有这个关中断操作的，现在最新的3.0.4在169行也是有关中断操作的。 [image]

嗯嗯，确认了下，没有问题，是我系统别的地方出问题，感谢！

closed 2023-04-10 by CornerOfSkyline