RT-Thread/rt-thread #7381: IAR __LDREX 和 __STREX 报 declared implicitly 警告问题

Zhyolo, 2023-04-26:
rt-thread 版本：RT-Thread v5.0.0
芯片：HC32F4A0PITB
IDE：IAR 9.30.1.50054

报警告文件：rt-thread/libcpu/arm/common/atomic_arm.c
[image]

下面这儿多了个 volatile
[image]

BernardXiong, 2023-04-26:
volatile的问题，欢迎发PR。 IAR `__LDREX / __STREX` 不太了解，是否需要在IAR上有什么设置，或者不是这么写法的？

mysterywolf, 2023-04-26:
https://github.com/RT-Thread/rt-thread/pull/7393

mysterywolf, 2023-04-26:
感谢提交PR！

Zhyolo, 2023-04-27:
原子操作的问题 #7393 并没有解决， #7393 只解决了 volatile 多了一个的问题

wenwen2101, 2023-05-10:
可以把__LDREXW的定义改成这样：
#elif defined (__IAR_SYSTEMS_ICC__)     /* for IAR Compiler */
#include "iccarm_builtin.h"
_Pragma("inline=forced") __intrinsic rt_atomic_t __LDREXW(volatile rt_atomic_t *ptr)
{
    return __iar_builtin_LDREX((volatile unsigned int *)ptr);
}
警告消失，并且确保使用iar的内置ldrex实现，在iar 9.30.1上编译通过

mysterywolf, 2023-09-20:
https://github.com/RT-Thread/rt-thread/pull/8060 解决

closed 2023-09-20 by mysterywolf