RT-Thread/rt-thread #7384: [HUST CSE]括号匹配错误

fly1ngpengu1ns, 2023-04-26:
在https://github.com/RT-Thread/rt-thread/blob/master/bsp/raspberry-pico/libraries/pico-sdk/src/rp2_common/pico_double/double_math.c#L328 这个文件中，存在下面这样的函数定义：
[image]
对于328和329行的函数，它们似乎括号并没有匹配？
是否应该由
```
double WRAPPER_FUNC(expm1(double x) { check_nan_d1(x); return exp)(x)-1; }
double WRAPPER_FUNC(log1p(double x) { check_nan_d1(x); return log)(1+x); }
```
变成
```
double WRAPPER_FUNC(expm1)(double x) { check_nan_d1(x); return exp(x)-1; }
double WRAPPER_FUNC(log1p)(double x) { check_nan_d1(x); return log(1+x); }
```

mysterywolf, 2023-04-26:
该源码非rt-thread源码 请向树莓派上游官方报告其HAL有缺陷，我们这里即便改了，下次他们版本更新，你改的内容也会被覆盖掉。

mysterywolf, 2023-04-26:
https://github.com/raspberrypi/pico-sdk

fly1ngpengu1ns, 2023-04-26:
好的，感谢您的审核

closed 2023-04-26 by fly1ngpengu1ns