RT-Thread/rt-thread #7386: [HUST CSE]使用了fopen后未使用fclose

fly1ngpengu1ns, 2023-04-26:
在https://github.com/RT-Thread/rt-thread/blob/master/examples/utest/testcases/posix/stdio_h/functions/perror_tc.c#L12 这个函数中，使用了fopen函数，却没有使用fclose函数
[image]
另外，在接下来的第18行`uassert_int_equal(perror_entry(), 0);`语句中，将perror_entry函数的返回值与0进行比较。但问题在于此函数只有一个返回值0，并没有其它返回值，因此是否需要将perror_entry函数修改成如下的形式？
```
static int perror_entry(void)
{
    FILE *stream;
    stream = fopen( "nulltest.txt", "r" );
    if(stream == NULL)
    {
        printf("perror test:");
        perror("nulltest.txt");
        return -1;
    }
    fclose(stream);
    return 0;
}
```

fly1ngpengu1ns, 2023-04-29:
> 嗯 确实有些问题 欢迎提交PR！

感谢您的支持，我们已经在 #7410 中修改了这个问题

mysterywolf, 2023-04-29:
感谢提交PR  已经合并

closed 2023-04-29 by mysterywolf