RT-Thread/rt-thread #7426: 新手任务

mysterywolf, 2023-04-29:
RT-Thread是一个基于社区的开源实时操作系统。RT-Thread代码的不断迭代和发展离不开社区小伙伴的积极贡献和提交代码。

但是一些社区小伙伴可能由于对RT-Thread代码不了解而导致不知道怎么向RT-Thread仓库提交代码。这里给大家做一个简单介绍。

RT-Thread主仓库就是你目前看到的这个仓库：https://github.com/RT-Thread/rt-thread， 我们主要在github下面维护和合并代码。同时，gitee也有同步镜像源，方便网络不通畅的小伙伴访问：https://gitee.com/rtthread/rt-thread

如果想要提交PR，在网络允许的情况下，尽量向RT-Thread github主仓库提交PR。如果网络条件不允许，也可以向gitee仓库提交PR，在review（视检）没有问题后，我们在将其同步到github下面并合并。

如果你刚开始接触RT-Thread，对RT-Thread代码还没有一个清晰的了解，但是非常想要为RT-Thread贡献代码，不妨尝试一下如下的**新手任务**。新手任务不会触及到核心代码的修改，你可以放心大胆的去写，并且在完成新手任务的过程中，你可以充分了解git相关知识、如何提交PR、以及RT-Thread代码风格和PR提交流程等相关基本内容。

如果你是新手，请按照以下顺序阅读相关视频、文档资料，并尝试下面的新手任务。

1. https://www.bilibili.com/video/BV1gr4y1w7yX 该视频教你怎么避免复杂的git命令行，基于图形化git来提交PR
2. https://github.com/RT-Thread/rt-thread/pull/5890 该PR给出了一个简单的新手PR模板

基本上阅读完上面两个资料，你给RT-Thread提交一个简单的PR就没有什么问题啦，后面的内容其实通过边提交PR就可以边学习到，另外如果有空也可以阅读一下[文档中心关系提交PR的内容](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/development-guide/github/github)

以下是新手任务，欢迎提交PR~

mysterywolf, 2023-04-29:
## 函数注释

目前，RT-Thread 内核函数的注释非常详细，例如：https://github.com/RT-Thread/rt-thread/blob/master/src/ipc.c#L308 ， 用户在编程时可以不用翻手册，直接找到函数定义，就可以知道这个函数的每个参数是什么含义，可以填哪几个参数，需要注意什么等等。

但是，RT-Thread的组件代码函数注释仍然欠缺与上面内核函数一样的详细注释，欢迎小伙伴参照上面给出的模板，对以下路径（文件夹以及其下属子文件夹）的.c源文件进行增加注释。

1. LIBC

路径：https://github.com/RT-Thread/rt-thread/tree/master/components/libc

LIBC文件夹内包含有标准LIBC、POSIX函数的一些实现，这些函数的定义可以直接在Google、百度、必应一搜就能搜到，非常容易增加函数注释，欢迎来补充注释

mysterywolf, 2023-04-29:
## 2 文档中心修改

文档中心：https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/README

RT-Thread的文档中心是一个可以由任何人编辑的文档，该文档中心在gitee仓库中维护，如果你在文档中心中发现任何已经过时的内容、汉语语句不通、汉语语法问题等等，哪怕是一个标点符号，都可以到Gitee仓库下来提交PR。

这里有一点需要注意，RT-Thread源码是在github下维护的，但是文档中心的仓库是在gitee下维护的，不要被绕蒙了😄

贡献参考：https://www.rt-thread.org/document/site/#/other/pr-rule/pr-rule
文档中心仓库地址：https://gitee.com/rtthread/docs-online

supperthomas, 2023-05-07:
https://github.com/RT-Thread/rt-thread/contribute

shu-qian, 2023-10-31:
I have submitted a PR to add comments. Could you help me check if there are any errors so that I can correct them in time?

PR #8209 (2023-11-01): Add comments to the rt-thread\components\libc\compilers\common\cstdlib.c
[
The purpose of this PR is to submit the  PR to Fixes #7426 
and add comments in the source code to improve the readability and understandability of the code and contribute to the project.

I have added detailed comments to rt-thread\components\libc\compilers\common\cstdlib.c

This PR is mainly for comments on the source code and does not involve specific functional changes, so there is no specific testing environment.
]

 - [ ] 本拉取/合并请求是一个草稿版本 This PR is for a code-review and is intended to get feedback
- [x] 本拉取/合并请求是一 [... cut]