RT-Thread/rt-thread #7452: NXP RT 系列芯片 keil 工程工具链版本问题

xym-ee, 2023-05-06:
## 问题描述

在学习 bsp 制作时我使用了野火 rt1021 开发板，基于仓库里已有的 bsp [imxrt1021-nxp-evk](https://github.com/RT-Thread/rt-thread/tree/master/bsp/imxrt/imxrt1021-nxp-evk) 进行修改，发现此 bsp 关闭所有组件、重新生成 keil 工程任然无法通过编译，提示启动文件有问题，随后发现 #6455 中也遇到此问题。

经过尝试，找到了问题原因：仓库中 rt1021 芯片使用了 gnu 汇编语法的启动文件，但是此工程基于 AC5 建立，无法处理 gnu 汇编语法的启动文件，因此提示以下错误信息

```
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(1): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(2): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(3): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(4): error: A1163E: Unknown opcode MIMXRT1021 , expecting opcode or Macro
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(5): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(6): error: A1167E: Invalid line start 
```

替换为 ARM 汇编语法的启动文件即可通过编译(随后提交 PR)，并在野火 rt1021 开发板上测试通过。

## 由此问题带出的一些想法

目前仓库中 rt 芯片 sdk 中 keil 工程使用的工具版本：

|SDK|工具链版本|startup_xxxx.s|
|-|-|-|
|所有1050系列的sdk          |AC5| arm 语法汇编|
|imxrt1060-nxp-evk(rt1062) |AC6| gnu 语法汇编|
|imxrt1064-nxp-evk         |AC5| arm 语法汇编|
|imxrt1170-nxp-evk         |AC6| gnu 语法汇编|

RT 系列芯片 NXP 官方 SDK 里 最近的使用 AC5 的 SDK 版本为 2.6.x，发布时间为 2019 年下半年。此后 keil 工程使用的都是 AC6，启动文件也由以前的 arm 汇编语法变成了 gnu 汇编语法。

是否考虑将 rt 系列芯片 bsp 中 keil 工程更新为使用 AC6？还是说只要能用，就先不动他。

BernardXiong, 2023-05-07:
这里的汇编文件应该需要分不同的编译器版本的

xym-ee, 2023-05-07:
使用这篇文章【[RT-Thread V4.1.0 新特性介绍之ArmClang](https://mp.weixin.qq.com/s/12EUP8kJaldd9elTJZG23Q)】里的方法来控制使用编译器版本嘛？

即修改 'rtconfig.py' 文件里的 'PLATFORM' 参数

mysterywolf, 2023-05-11:
已经和NXP的小伙伴说了这个问题，会在下周左右回复

ccx-lan, 2023-05-15:
还有就是rt系列芯片调试和下载经常会使用不同的分散加载文件，现有的bsp都是nor版本的，而要在rtt scons上添加多种分散加载的mdk版本也是麻烦，像下面这种，我是搞两个工程，只能时不时同步一下文件，希望NXP技术人员还是哪位大佬能搞一下
[image]

mysterywolf, 2023-05-17:
关于1020板子AC6更新的pr，不用合并，原来用的就是可以的，但是需要修改一些编译配置，NXP小伙伴会整理然后更新到AC6

1050和1064 不打算更新到AC6, 这两块板子很多外设可以和1060公用 所以没有更新打算

mysterywolf, 2023-05-17:
https://github.com/RT-Thread/rt-thread/pull/7518

closed 2023-05-19 by mysterywolf