RT-Thread/rt-thread #7466: rt_device_open/close对device_open的处理不对称

labrick, 2023-05-09:
现象：在RTT中，多次open/close同一个device，open会多次调用driver，但close只会调用一次
芯片/BSP：STM32
工具链：arm-non-eabi-
RT-Thread版本: 4.0.1

复现步骤：
1. driver中open/close打印信息
2. 构建多线程，每个线程中都open同一个driver，然后close
3. check log，发现open会有多次，close只会有一次

原因：
```C
rt_err_t rt_device_open(rt_device_t dev, rt_uint16_t oflag)
{
    rt_err_t result = RT_EOK;
    ....
    /* call device_open interface */
    if (device_open != RT_NULL)
    {
        result = device_open(dev, oflag);
    }
    else
    {
        /* set open flag */
        dev->open_flag = (oflag & RT_DEVICE_OFLAG_MASK);
    }

    /* set open flag */
    if (result == RT_EOK || result == -RT_ENOSYS)
    {
        dev->open_flag |= RT_DEVICE_OFLAG_OPEN;

        dev->ref_count++;
        /* don't let bad things happen silently. If you are bitten by this assert,
         * please set the ref_count to a bigger type. */
        RT_ASSERT(dev->ref_count != 0);
    }

    return result;
}
RTM_EXPORT(rt_device_open);

rt_err_t rt_device_close(rt_device_t dev)
{
    rt_err_t result = RT_EOK;

    RT_ASSERT(dev != RT_NULL);
    RT_ASSERT(rt_object_get_type(&dev->parent) == RT_Object_Class_Device);

    if (dev->ref_count == 0)
        return -RT_ERROR;

    dev->ref_count--;

    if (dev->ref_count != 0)
        return RT_EOK;

    /* call device_close interface */
    if (device_close != RT_NULL)
    {
        result = device_close(dev);
    }

    /* set open flag */
    if (result == RT_EOK || result == -RT_ENOSYS)
        dev->open_flag = RT_DEVICE_OFLAG_CLOSE;

    return result;
}
RTM_EXPORT(rt_device_close);
```

从上面的代码看
1. open先调用device_open，然后再ref_count++，也就是基本都会调用device_open
2. close先判断ref_count，如果ref_count > 0直接返回，而不再调用device_close

Guozhanxin, 2023-05-10:
框架就是这么设计的，因为不同的设备，对于多次打开的行为可能不一样，所以具体的设备要在open的ops里对打开次数ref_count做一下判断，如果设备支持多次打开，并且都需要进行操作，就真正的执行打开的操作，否则直接返回就好了。

labrick, 2023-05-10:
> 框架就是这么设计的，因为不同的设备，对于多次打开的行为可能不一样
从linux角度看，同一个进程，打开同一个设备，driver只会在第一次得到调用，后面都会被文件系统拦截下来，只是返回duplicate的fd，这样做的原因，同一个进程资源共享，打开一次就已经足够，多次打开的用途是什么呢？

> 所以具体的设备要在open的ops里对打开次数ref_count做一下判断，如果设备支持多次打开，并且都需要进行操作，就真正的执行打开的操作，否则直接返回就好了。
 driver需要主动拿到dev->ref_cnt的值做判断，理论上driver不应该碰内核数据结构的成员吧，因为这是一个非常危险的行为

而且这个设计会导致框架层次：open/close在不再对称
这个设计是经过讨论之后的结论吗？