RT-Thread/rt-thread #7471: RT-Thread v4.1.1 CAN驱动接收中断在两个设备互发消息时热插拔会导致系统卡死

logeexpluoqi, 2023-05-10:
在版本v4.1.1中使用can_drv驱动，没有使用滤波器功能，基于STM32F407芯片，两个设备会互相发心跳，频率为1s，其间可能会以200Hz频率发送一些数据。当把CAN的连线给断开后重新插上后，极大可能性会导致其中一个设备卡死，msh无法工作，有大概一半可能性会导致两个设备同时卡死。
在经过排查后，发现注释掉can设备初始化设置中的几个中断允许函数后问题解决，如下：
```
__HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO0_MSG_PENDING);
 // __HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO0_FULL);
 // __HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO0_OVERRUN);
__HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO1_MSG_PENDING);
  // __HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO1_FULL);
  // __HAL_CAN_ENABLE_IT(&drv_can->CanHandle, CAN_IT_RX_FIFO1_OVERRUN);
```
注释后插拔测试中，没有发现设备卡死情况

wdfk-prog, 2023-05-11:
- 这两个是错误中断啊.你进了错误了啊.
- 你屏蔽了错误而已.错误还是发生了.
- 不应该这边注释,而是错误中断中去处理

logeexpluoqi, 2023-05-11:
嗷嗷嗷，是这样。我去改一下

closed 2023-05-11 by logeexpluoqi