RT-Thread/rt-thread #7485: error: 'PATH_MAX' and 'NAME_MAX' undeclared (first use in this function)

faroperfect, 2023-05-12:
按照文档所示，使用QEMU模拟器，编译bsp时，执行scons编译，出现error
huron@huron-T90:[~/rt-thread/bsp/qemu-vexpress-a9]:13:44
$scons
scons: Reading SConscript files ...
Newlib version: 3.3.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/kernel/components/libc/posix/ipc/mqueue.o
/home/huron/rt-thread/components/libc/posix/ipc/mqueue.c: In function 'mq_open':
/home/huron/rt-thread/components/libc/posix/ipc/mqueue.c:141:15: error: 'PATH_MAX' undeclared (first use in this function)
  141 |     if (len > PATH_MAX || len > NAME_MAX)
      |               ^~~~~~~~
/home/huron/rt-thread/components/libc/posix/ipc/mqueue.c:141:15: note: each undeclared identifier is reported only once for each function it appears in
/home/huron/rt-thread/components/libc/posix/ipc/mqueue.c:141:33: error: 'NAME_MAX' undeclared (first use in this function); did you mean 'RT_NAME_MAX'?
  141 |     if (len > PATH_MAX || len > NAME_MAX)
      |                                 ^~~~~~~~
      |                                 RT_NAME_MAX
scons: *** [build/kernel/components/libc/posix/ipc/mqueue.o] Error 1
scons: building terminated because of errors.
在此之前执行了source。

DylanLiuH2O, 2023-05-12:
我也遇到了
```text
...
CC build/kernel/components/libc/posix/io/poll/select.o
CC build/kernel/components/libc/posix/io/stdio/libc.o
CC build/kernel/components/libc/posix/io/termios/termios.o
CC build/kernel/components/libc/posix/ipc/mqueue.o
/home/dylan/workspace/repos/rt-thread/components/libc/posix/ipc/mqueue.c: In function 'mq_open':
/home/dylan/workspace/repos/rt-thread/components/libc/posix/ipc/mqueue.c:141:33: error: 'NAME_MAX' undeclared (first use in this function)
     if (len > PATH_MAX || len > NAME_MAX)
                                 ^~~~~~~~
/home/dylan/workspace/repos/rt-thread/components/libc/posix/ipc/mqueue.c:141:33: note: each undeclared identifier is reported only once for each function it appears in
scons: *** [build/kernel/components/libc/posix/ipc/mqueue.o] Error 1
scons: building terminated because of errors.
```

DylanLiuH2O, 2023-05-14:
> 按照文档所示，使用QEMU模拟器，编译bsp时，执行scons编译，出现error
> huron@huron-T90:[~/rt-thread/bsp/qemu-vexpress-a9]:13:44
> $scons
> scons: Reading SConscript files ...
> Newlib version: 3.3.0
> scons: done reading SConscript [...]

你在哪个平台进行的编译？`PATH_NAME`好像是可选的定义，并不强制要求系统定义这个宏。

supperthomas, 2023-05-22:
error: 'PATH_MAX' undeclared

closed 2023-05-23 by mysterywolf