RT-Thread/rt-thread #7491: 4.1.0 下 qemu-vexpress-a9 编译报错 error: unknown type name 'fd_mask'

IcyFeather233, 2023-05-13:
# 环境

Win 10
env v1.3.5
RT-Thread v4.1.0

# 复现

进入 rt-thread-4.1.0\bsp\qemu-vexpress-a9 后 scons 编译报错 error: unknown type name 'fd_mask'：

```
C:\Users\IcyFeather\Downloads\rt-thread-4.1.0\components\libc\posix\io\poll\select.c: In function 'fdszero':
C:\Users\IcyFeather\Downloads\rt-thread-4.1.0\components\libc\posix\io\poll\select.c:17:5: error: unknown type name 'fd_mask'
   17 |     fd_mask *m;
      |     ^~~~~~~
C:\Users\IcyFeather\Downloads\rt-thread-4.1.0\components\libc\posix\io\poll\select.c:24:10: error: 'fd_mask' undeclared (first use in this function)
   24 |     m = (fd_mask *)set;
      |          ^~~~~~~
C:\Users\IcyFeather\Downloads\rt-thread-4.1.0\components\libc\posix\io\poll\select.c:24:10: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\IcyFeather\Downloads\rt-thread-4.1.0\components\libc\posix\io\poll\select.c:24:19: error: expected expression before ')' token
   24 |     m = (fd_mask *)set;
      |                   ^
scons: *** [build\kernel\components\libc\posix\io\poll\select.o] Error 1
scons: building terminated because of errors.
```

IcyFeather233, 2023-05-13:
补充：v4.0.5 也有一样的问题

mysterywolf, 2023-05-23:
验证了一下 最新代码没有这个问题

mysterywolf, 2023-05-23:
[image]

mysterywolf, 2023-05-23:
https://github.com/RT-Thread/rt-thread/pull/5790 这个问题已经在这里得到修复了

closed 2023-05-23 by mysterywolf