RT-Thread/rt-thread #7498: atomic，原子操作的问题

BernardXiong, 2023-05-14:
原子操作是现代计算机中十分重要的功能，在多核场景下是必备的功能之一，也需要成为RT-Thread系统最基本的功能。

目前存在的问题：

* `atomic_compare_exchange_strong` 声明，函数参数不一致
  * rtatomic.h 中的定义 `rt_atomic_t rt_hw_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr, rt_atomic_t *old, rt_atomic_t new);`
  * sw_atomic 中的定义`rt_atomic_t rt_soft_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr1, rt_atomic_t *ptr2,
        rt_atomic_t desired);`
  * libcpu/arm 中的定义`rt_atomic_t rt_hw_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr, volatile rt_atomic_t *old, rt_atomic_t new);`
  * libcpu/risc-v 中的定义`rt_atomic_t rt_hw_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr, volatile rt_atomic_t *old, rt_atomic_t new);`
  * 最终导致编译错误：
```bash
CC build/kernel/libcpu/arm/common/atomic_arm.o
/home/bernard/workspace/rt-thread/libcpu/arm/common/atomic_arm.c:169:13: error: conflicting types for 'rt_hw_atomic_compare_exchange_strong'
  169 | rt_atomic_t rt_hw_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr, volatile rt_atomic_t *old, rt_atomic_t new)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/bernard/workspace/rt-thread/include/rtthread.h:30,
                 from /home/bernard/workspace/rt-thread/libcpu/arm/common/atomic_arm.c:11:
/home/bernard/workspace/rt-thread/include/rtatomic.h:23:13: note: previous declaration of 'rt_hw_atomic_compare_exchange_strong' was here
   23 | rt_atomic_t rt_hw_atomic_compare_exchange_strong(volatile rt_atomic_t *ptr, rt_atomic_t *old, rt_atomic_t new);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
* 在整体系统中需要做到在任意情况下都可以使用原子操作（必定能够在std_atomic，hw_atomic，sw_atomic三者中选择一个）
  * 如果定义了hw_atomic，则优先使用；
  * 如果定义了std_atomic则使用，或者自动探测是否使用stdatomic；
  * 最后，确定是否使用sw_atomic

Guozhanxin, 2023-05-20:
什么编译器报错的啊，我用gcc没报错

BernardXiong, 2023-05-20:
这块问题还挺多的

```bash
[cpp_check.py 28 INFO] libcpu/risc-v/common/atomic_riscv.c:21:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:32:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:44:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:55:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:66:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:77:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:88:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:110:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
libcpu/risc-v/common/atomic_riscv.c:158:12: error: Uninitialized variable: result [uninitvar]
    return result;
           ^
```

byte-iot, 2023-06-28:
问题解决了吗，我也遇到这个问题了

closed 2023-07-26 by mysterywolf