RT-Thread/rt-thread #7509: HUST Check List

mysterywolf, 2023-05-15:


mysterywolf, 2023-05-15:
## 新手任务
- https://github.com/RT-Thread/rt-thread/issues/7426
- https://github.com/RT-Thread/rt-thread/issues/7145
- https://github.com/RT-Thread/rt-thread/issues?q=is%3Aissue+is%3Aopen+label%3ADoc

sexenteen, 2026-11-01:
这个 issue 看起来可以作为「活动清单」归档了，我先做一个核实：

**1）它本身不是缺陷，是一次扫描活动的邀请清单**

正文是「欢迎华中科技大学 IoTS&P 实验室的同学们着重扫描以下仓库」，列出了主仓 + 8 个软件包仓；
评论里是 @mysterywolf 给的「新手任务」索引（#7426、#7145、Doc 标签、gitee docs-online）。
两者都不是本 issue 待完成的技术交付物 —— 也就是说它没有「做完」的判据，只有「活动结束」的时点。

**2）活动期已过，产出也都闭环了**

- 该标签（`University: HUST IoTS&P Lab`）下共 **96 条** issue：**95 条已关闭，只有本 issue 还是 open**，
  且时间集中在 2023-04 ~ 2023-06。
- 据此清单提交的 PR #7573（`pico_malloc.c`）在 2023-05-26 当天被关闭（未合并）。
- 本 issue 自 2023-05-15 创建后再无更新（已 3 年多）。
- 评论里引用的任务：#7145 已关闭。#7426（「新手任务 | 提交PR」）仍是 open，但它是独立的通用新手任务列表，
  不依赖本 issue 存续。

**3）建议**

- 如果这项高校合作已经结束，建议**归档关闭**；
- 如果后续还会继续组织同学做代码清扫，那保留它作为入口也是合理的。

@mysterywolf 麻烦确认一下这个清单后续是否还会使用，谢谢！