RT-Thread/rt-thread #7533: [libc error]  编译错误PATH_MAX 未定义

supperthomas, 2023-05-19:
任意STM32 bsp
stm32\stm32l496-st-nucleo
menuconfig选中mqueue.h
如下图
[image]
之后执行`scons --target=mdk5`
会出现编译错误：
[image]

mysterywolf, 2023-05-20:
这块写的有问题，周一我看一下

supperthomas, 2023-05-22:
@DylanLiuH2O

mysterywolf, 2023-05-22:
涛哥 你这个先不用着急 可以直接把PATH_MAX这个判断去掉，这个判断的逻辑写的有问题，加>PATH_MAX这个判断完全没必要

DylanLiuH2O, 2023-05-22:
> 涛哥 你这个先不用着急 可以直接把PATH_MAX这个判断去掉，这个判断的逻辑写的有问题，加>PATH_MAX这个判断完全没必要 我得联系谁写的这个，谁来改一下

原始作者的提交在这里#7395 ，`PATH_MAX`是一个平台相关的宏，而且并不是强制性的，因此可能会报未定义。
> Macro: int PATH_MAX
> The uniform system limit (if any) for the length of an entire file name (that is, the argument given to system calls such as open), including the terminating null character.
> **Portability Note: The GNU C Library does not enforce this limit even if PATH_MAX is defined.**

[Limits-for-Files](https://www.gnu.org/software/libc/manual/html_node/Limits-for-Files.html)

mysterywolf, 2023-05-23:
fixed in here: https://github.com/RT-Thread/rt-thread/pull/7550

closed 2023-05-23 by mysterywolf