RT-Thread/rt-thread #7551: rtatomic.h头文件中对函数”rt_hw_interrupt_disable“的引用未声明

hidflower, 2023-05-23:
在编译V5.0.0的内核文件如clock.c时，出现警告：include\rtatomic.h(88)：function “rt_hw_interrupt_disable” declared implicitly。
1. 分析clock.c相关头文件包含顺序及内容，如下：
（1） clock.c中：
020  #include <rthw.h>

（2）rthw.h中：
021  #include <rtthread.h>
       ...
123  #ifdef RT_USING_SMP
124  rt_base_t rt_hw_local_irq_disable();
125  void rt_hw_local_irq_enable(rt_base_t level);
126
127  #define rt_hw_interrupt_disable rt_cpus_lock
128  #define rt_hw_interrupt_enable rt_cpus_unlock
129 
130  #else
131  **rt_base_t rt_hw_interrupt_disable(void);**
132  void rt_hw_interrupt_enable(rt_base_t level);
131  #endif /*RT_USING_SMP*/

（3）rtthread.h中：
031   #include <rtatomic.h>

（4）rtatomic.h中：
067  #include <rthw.h>
        ...
084  rt_inline rt_atomic_t rt_soft_atomic_exchange(volatile rt_atomic_t *ptr, rt_atomic_t val)
085  {
086      rt_base_t level;
087      rt_atomic_t temp;
088      **level = rt_hw_interrupt_disable();**
089      temp = *ptr;
090     *ptr = val;
091      rt_hw_interrupt_enable(level);
092      return temp;
093  }

2. 由上述内容可见：
（1）rthw.h中，函数“rt_hw_interrupt_disable“的声明是在#include <rtthread.h>之后，所以rtthread.h中所包含的rtatomic.h中引用“rt_hw_interrupt_disable“函数时，该函数尚未声明。
（2）虽然rtatomic.h中在引用“rt_hw_interrupt_disable“函数前，增加了#include  <rthw.h>，但相当于是rthw.h包含自身，是无效的包含。

zhkag, 2023-06-09:
#7639

BernardXiong, 2023-06-10:
Fixed it with #7639.

Please re-check this issue, thanks.

closed 2023-06-10 by mysterywolf

hidflower, 2023-06-12:
> Fixed it with #7639.
> 
> Please re-check this issue, thanks.

按#7639的方法能够解决这个问题，谢谢