RT-Thread/rt-thread #7554: armclang 和 llvm-arm 的 platform 是不是统一成 gcc。

Guozhanxin, 2023-05-23:
发现他们在源码层面上是一样的。不同点可能是支持的编译参数不太一样。
[image]

感觉可能的方案是，统一平台为 gcc，然后再不同的地方，再根据交叉工具链 CROSS_TOOL 字段来区别判断
  https://github.com/RT-Thread/rt-thread/pull/7553/files

mysterywolf, 2023-05-23:
CROSS_TOOL 这个字段基本没啥太大作用 他是判断IDE类型的，例如AC5 AC6的CROSS_TOOL 都叫做 keil 但是PLATFORM的名字一个叫armcc 一个叫armclang，区分都在PLATFORM这里

mysterywolf, 2023-05-23:
llvm-arm这块先别扩大bsp支持范围，就先苟在潘多拉板子上就行？等试验稳定了，再推广的其他bsp上

mysterywolf, 2023-05-23:
可以把CROSS_TOOL 设置为 gcc

mysterywolf, 2023-05-23:
其实我早就发现这个问题了，感觉我们的CROSS_TOOL和PLATFORM一直是用反了，但是已经这样 就没有再拧过来

mysterywolf, 2023-05-23:
CROSS_TOOL 可以都设置为 gcc

mysterywolf, 2023-05-23:
我觉得 https://github.com/RT-Thread/rt-thread/pull/7553 就按照这个PR这么改没有问题。增加新的工具链支持的时候手动确认是否支持，llvm armclang只不过是兼容gcc语法。直接按照gcc处理还是有些粗暴还容易埋坑。手动确认编译没问题，再针对某个特定bsp增加支持。

mysterywolf, 2023-05-25:
或者是封装一个函数 is_gcc_like

closed 2023-08-21 by mysterywolf