RT-Thread/rt-thread #7565: [CI] kernel/atomic_c11  error

supperthomas, 2023-05-25:
https://github.com/RT-Thread/rt-thread/actions/workflows/action_utest.yml
[image]

mysterywolf, 2023-05-25:
https://github.com/RT-Thread/rt-thread/commit/0315c0dee76ce67e3659744d4d280659e0a13906 从这个PR开始报错的 但是这个PR不应该影响原子操作才对...

supperthomas, 2023-05-26:
又好了

supperthomas, 2023-05-29:
@dejavudwh 有空可以看下啥问题。

dejavudwh, 2023-05-29:
> @dejavudwh 有空可以看下啥问题。

好的

supperthomas, 2023-05-31:
```
[D/utest] [    OK    ] [ unit     ] (test_atomic_api:120) is passed
[D/utest] [    OK    ] [ unit     ] (test_atomic_api:121) is passed
[E/utest] [  ASSERT  ] [ unit     ] at (atomic_tc.c); func: (test_atomic_add:152); msg: ((count == 3000000) is false)
[E/utest] [  FAILED  ] [ result   ] testcase (testcases.kernel.atomic_tc)
[I/utest] [----------] [ testcase ] (testcases.kernel.atomic_tc) finished
[I/utest] [----------] [ testcase ] (testcases.kernel.mem_tc) started
[D/utest] [    OK    ] [ unit     ] (mem_functional_test:94) is passed
```

supperthomas, 2023-05-31:
[exe.zip](https://github.com/RT-Thread/rt-thread/files/11607581/exe.zip)

ubuntu 工程

supperthomas, 2023-05-31:
@Guozhanxin

supperthomas, 2023-06-01:
怎么感觉这个CI 一会好，一会不好的？

closed 2023-06-10 by BernardXiong