RT-Thread/rt-thread #7587: [scons --dist] scons --dist 没有example目录

supperthomas, 2023-05-31:
[exe.zip](https://github.com/RT-Thread/rt-thread/files/11607581/exe.zip)

qemu-vexpress-a9 目录选中example中的utest之后，scons --dist 打包之后的工程中没有example目录
[image]

mysterywolf, 2023-05-31:
example目录是故意被排除掉的

supperthomas, 2023-06-01:
> example目录是故意被排除掉的

为什么？

mysterywolf, 2023-06-01:
example只是例程，不视为工程的一部分，需要参考可以直接翻仓库。而且example并没有sconscript控制

supperthomas, 2023-06-01:
> example只是例程，不视为工程的一部分，需要参考可以直接翻仓库。而且example并没有sconscript控制

没有可以加一下。

mysterywolf, 2023-06-01:
我个人不建议加，example里的代码都是一些老代码，格式风格很不好。我们正式的例程是在软件包里 分别有kernel device 网络等等。

mysterywolf, 2023-06-01:
@Guozhanxin

supperthomas, 2023-06-01:
utest应该是比较新的代码。如果出了问题，打包一个完整工程出来debug还是比较方便的。这样不需要把CI当时的所有bsp都拉出来。
应该尽量保证scons --dist 打包出来的工程和bsp当时的工程完全吻合~。

mysterywolf, 2023-06-01:
我在想 要不要把utest提出来到根目录，example里边的大杂烩该放到sample软件包的放软件包、没用的直接删掉。

closed 2023-06-26 by mysterywolf