RT-Thread/rt-thread #7592: [E/DBG] virtio-blk0 mount faile

TaiJuWu, 2023-06-01:
我使用RT-thread main branch
bsp為qemu-virt64-riscv
toolchain用riscv64-unknown-elf-
在執行rt-thread時會產生錯誤 [E/DBG] virtio-blk0 mount faile

截圖如下
[image]

heyuanjie87, 2023-06-11:
mkfs -t elm virtio-blk0 执行过吗

TaiJuWu, 2023-06-19:
現在好像正常執行了
沒辦法reproduce

closed 2023-06-19 by TaiJuWu

rstg00po54, 2023-11-30:
我也遇到了这个问题
 \ | /
- RT -     Thread Smart Operating System
 / | \     5.0.0 build Nov 30 2023
 2006 - 2020 Copyright by rt-thread team
lwIP-2.0.2 initialized!
[I/sal.skt] Socket Abstraction Layer initialize success.
[I/utest] utest is initialize success.
[I/utest] total utest testcase num: (1)
[E/DBG] virtio-blk0 mount failed

Hello RISC-V
msh />
msh />
msh />
msh />mkfs -t elm virtio-blk0 
msh />ls
No such directory
msh />dir
dir: command not found.
msh />

应该怎么解决呢