RT-Thread/rt-thread #7594: RISC-V multi-cpu porting AS error

TaiJuWu, 2023-06-02:
芯片: qemu riscv
BSP: qemu-virt64-riscv
RT-Thread v5.0.0
toolchain: riscv64-unknown-elf-gcc

error message:
AS build/kernel/libcpu/risc-v/virt64/context_gcc.o
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h: Assembler messages:
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:53: Error: unrecognized opcode `typedef union{'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:54: Error: unrecognized opcode `unsigned long slock'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:55: Error: unrecognized opcode `struct __arch_tickets{'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:56: Error: unrecognized opcode `unsigned short owner'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:57: Error: unrecognized opcode `unsigned short next'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:58: Error: junk at end of line, first unrecognized character is `}'
/home/tjwu1217/rt-thread/libcpu/risc-v/virt64/cpuport.h:59: Error: junk at end of line, first unrecognized character is `}'
scons: *** [build/kernel/libcpu/risc-v/virt64/context_gcc.o] Error 1

xqyjlj, 2023-07-18:
换成这个工具链再试试？ https://download.rt-thread.org/rt-smart/riscv64/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_180881.tar.bz2

yf13, 2023-09-01:
Why not put this toolchain URL in the "bsp/qemu-virt64-riscv/README.md" directly?
I've failed with toolchains in ubuntu repository (like gcc-riscv64-linux-gnu) and both 10.2 and 11 in  musl.cc.

Yohozzy, 2024-02-26:
遇到同样的问题 
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h: Assembler messages:
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:21: Error: unrecognized opcode `typedef union{'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:22: Error: unrecognized opcode `unsigned long slock'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:23: Error: unrecognized opcode `struct __arch_tickets{'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:24: Error: unrecognized opcode `unsigned short owner'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:25: Error: unrecognized opcode `unsigned short next'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:26: Error: junk at end of line, first unrecognized character is `}'
D:\RT-ThreadStudio\workspace\test\rt-thread\libcpu\risc-v\common/cpuport.h:27: Error: junk at end of line, first unrecognized character is `}'

芯片: HPM6750

Rbb666, 2024-07-08:
qemu-virt64-riscv 这个BSP不支持SMP吧

unicornx, 2024-12-17:
@Rbb666 我建议 close 这个 issue。

首先 virt64 riscv 不支持  smp。其次目前主线 smart 我验证过已经没有问题。参考 #9622

TaiJuWu, 2024-12-17:
@unicornx 感謝回報，因為現在我已經沒在用RT-thread，所以直接關閉這個issue

closed 2024-12-17 by TaiJuWu