RT-Thread/rt-thread #7607: FIXME:LPC4088支持编译环境问题

meng-plus, 2023-06-05:
1. 经测试，使用RTT官方开发环境，引入的gcc启动文件无法正常使用，自己尝试修改无果
2. env 生成IAR开发环境不支持，自己简单做了下能自己用，希望给出官方支持

mysterywolf, 2023-06-05:
这个bsp用的人不多，真正一次最后维护都是19年左右的了

mysterywolf, 2023-06-05:
欢迎提交PR 继续维护这个BSP

supperthomas, 2023-06-05:
[                - "lpc408x"](https://github.com/RT-Thread/rt-thread/blob/b9cd90c60084e8baf08d65a1f170eaae88463729/.github/workflows/action.yml#LL170C1-L170C1)

这个bsp MDK下面可以用吗？

supperthomas, 2023-06-05:
你可以在论坛上详细贴出你遇到的问题：https://club.rt-thread.org/index.html 。
这些bsp都是社区爱好者提供的，我们这边没有这类开发板，没办法验证跑起来的正确性。不过问题应该都差不多的，所以你详细描述问题，论坛提问，大家可以帮你解决。也欢迎PR 进行维护~

meng-plus, 2023-06-05:
> 欢迎提交PR 继续维护这个BSP

我提交了PT https://github.com/RT-Thread/rt-thread/pull/7611 本机测试运行通过 更多问题如有发现我再提交

meng-plus, 2023-06-05:
> [- “LPC408X”](https://github.com/RT-Thread/rt-thread/blob/b9cd90c60084e8baf08d65a1f170eaae88463729/.github/workflows/action.yml#LL170C1-L170C1)
> 
> 这个bsp MDK下面可以用吗？

我这里只用IAR环境，MDK也许能用，毕竟gcc环境的也不能跑，从NXP官网中我获知道LPC4088,只放出IAR、MDK两中环境的例程代码

supperthomas, 2023-06-05:
> > [- “LPC408X”](https://github.com/RT-Thread/rt-thread/blob/b9cd90c60084e8baf08d65a1f170eaae88463729/.github/workflows/action.yml#LL170C1-L170C1)
> > 
> > 这个bsp MDK下面可以用吗？
> 
> 我这里只用IAR环境，MDK也许能用，毕竟gcc环境的也不能跑，从NXP官网中我获知道LPC4088,只放出IAR、MDK两中环境的例程代码

感谢PR，可以自己下个MDK试试，帮忙维护一下这个BSP

closed 2023-06-06 by meng-plus