RT-Thread/rt-thread #7638: [CI] 讨论一下action.yml 中可以完善的点，欢迎吐槽

supperthomas, 2023-06-08:
目前action.yml 中对所有action中的bsp都做了scons检查，大家觉得，是不是提交的bsp没有对一些I2C 或者SPI 等做编译检查呀？会检查到吗？

dejavudwh, 2023-06-08:
1. 感觉需要仔细检查这个标准不太好界定。如果只是简单来考虑的话，就是更改了这个bsp相关的就只去重新编译这个bsp，如果修改了内核相关的就全部重新编译。
2. 查看驱动的编译情况感觉还是有必要的。

wdfk-prog, 2023-06-09:
能不能pr的人可以选择开启对应的CI检查.比如我改了SPI,可以开启SPI的选项去检查.

supperthomas, 2023-06-09:
> 能不能pr的人可以选择开启对应的CI检查.比如我改了SPI,可以开启SPI的选项去检查.

好想法

mysterywolf, 2023-06-09:
RTduino成熟之后，可以把关键的一些外设进行硬件在环看护。github服务器编译好的bin文件下发到本地服务器/宿主计算机，然后烧录到板子上。实现硬件在环看护之后，就可以对一些关键新能指标进行看护。例如SD的拷贝速度，SPI的传输速度等。

supperthomas, 2023-06-09:
> RTduino成熟之后，可以把关键的一些外设进行硬件在环看护。github服务器编译好的bin文件下发到本地服务器/宿主计算机，然后烧录到板子上。实现硬件在环看护之后，就可以对一些关键新能指标进行看护。例如SD的拷贝速度，SPI的传输速度等。
> 
> 上面提到的I2C, SPI, PWM , ADC, DAC 这些外设，RTduino都有简单统一的API，验证更方便些。

仓库链接提供一下？

mysterywolf, 2023-06-09:
https://github.com/RTduino/RTduino

mysterywolf, 2023-06-14:
https://github.com/RT-Thread/rt-thread/issues/7510 把这个加上吧

supperthomas, 2024-10-07:
> 能不能pr的人可以选择开启对应的CI检查.比如我改了SPI,可以开启SPI的选项去检查.

其实目前可以到自己的RT-THREAD仓库查看manual_scons_dist_trigger_only_one  自行手动选择config进行编译，但问题是，很少会主动自己编译

[image]

supperthomas, 2025-02-26:
- [ ] pkgs --update => pkgs --update-force

supperthomas, 2025-02-27:
- [ ] 如果只提交bsp相关的，判断当前修改的bsp目录，如果只有bsp 比如`stm32/stm32l496-st-nucleo` ci中，可以匹配一下。其他bsp不进行检查