RT-Thread/rt-thread #7658: 当不使用RT_USING_LIBC时，rtthread.h中使用的 va_list 未定义

hidflower, 2023-06-12:
在V5.0.0中，当不使用RT_USING_LIBC时，编译内核文件如clock.c时，出现错误：include\rtthread.h(643): error: #20: identifier "va_list" is undefined

1. 检查报错位置代码如下：
int rt_vsprintf(char *dest, const char *format, **va_list** arg_ptr);

2. va_list是在<stdarg.h>中定义的，根据rtdef.h的内容，可知RTTHREAD内核设计只有在使用RT_USING_LIBC时，才允许使用va_list。如下：
**#ifdef RT_USING_LIBC**
#include <stdint.h>
#include <stddef.h>
**#include <stdarg.h>**
#include <sys/types.h>
**#endif** /* RT_USING_LIBC */

3. 但是，在rtthread.h中，声明rt_vsprintf等函数时，未进行RT_USING_LIBC的检查。建议增加RT_USING_LIBC检查，如下：
**#ifdef RT_USING_LIBC**	
int rt_vsprintf(char *dest, const char *format, va_list arg_ptr);
int rt_vsnprintf(char *buf, rt_size_t size, const char *fmt, va_list args);
**#endif**

mysterywolf, 2023-06-12:
感谢建议！

mysterywolf, 2023-10-04:
考虑5.0.3将klibc.h/.c拆出来之后 处理此事