RT-Thread/rt-thread #7707: [NEON]在armv8-a架构下，任务切换时，v0-v31向量寄存器是否进行了现场保存

xielizyh, 2023-06-21:
在asm_fpu.h下只有Q0-Q15的寄存器保存和恢复操作，是否还需要对向量寄存器v0-v31进行保存，谢谢？
.macro SAVE_FPU, reg
    STR Q0, [\reg, #-0x10]!
    STR Q1, [\reg, #-0x10]!
    STR Q2, [\reg, #-0x10]!
    STR Q3, [\reg, #-0x10]!
    STR Q4, [\reg, #-0x10]!
    STR Q5, [\reg, #-0x10]!
    STR Q6, [\reg, #-0x10]!
    STR Q7, [\reg, #-0x10]!
    STR Q8, [\reg, #-0x10]!
    STR Q9, [\reg, #-0x10]!
    STR Q10, [\reg, #-0x10]!
    STR Q11, [\reg, #-0x10]!
    STR Q12, [\reg, #-0x10]!
    STR Q13, [\reg, #-0x10]!
    STR Q14, [\reg, #-0x10]!
    STR Q15, [\reg, #-0x10]!
.endm

.macro RESTORE_FPU, reg
    LDR Q15, [\reg], #0x10
    LDR Q14, [\reg], #0x10
    LDR Q13, [\reg], #0x10
    LDR Q12, [\reg], #0x10
    LDR Q11, [\reg], #0x10
    LDR Q10, [\reg], #0x10
    LDR Q9, [\reg], #0x10
    LDR Q8, [\reg], #0x10
    LDR Q7, [\reg], #0x10
    LDR Q6, [\reg], #0x10
    LDR Q5, [\reg], #0x10
    LDR Q4, [\reg], #0x10
    LDR Q3, [\reg], #0x10
    LDR Q2, [\reg], #0x10
    LDR Q1, [\reg], #0x10
    LDR Q0, [\reg], #0x10
.endm

aozima, 2023-06-21:
如果他们是独立的寄存器，就需要保存。
如果是同个物理寄存器在不同模式下的别名，就不需要。
类似的还有单精度模式下的`S0~S31`等于双精度下的`D0~D15`

xielizyh, 2023-06-27:
> 如果他们是独立的寄存器，就需要保存。 如果是同个物理寄存器在不同模式下的别名，就不需要。 类似的还有单精度模式下的`S0~S31`等于双精度下的`D0~D15`

V0-V15相当于Q0-Q15，但是V16-V31没有进行保存，测试发现如果添加Q16保存，系统可以正常，但是添加多添加一个Q17保存，系统会奔溃，官方可以支持一下对armv8-a向量寄存器的保存吗？

GuEe-GUI, 2023-07-10:
> > 如果他们是独立的寄存器，就需要保存。 如果是同个物理寄存器在不同模式下的别名，就不需要。 类似的还有单精度模式下的`S0~S31`等于双精度下的`D0~D15`
> 
> V0-V15相当于Q0-Q15，但是V16-V31没有进行保存，测试发现如果添加Q16保存，系统可以正常，但是添加多添加一个Q17保存，系统会奔溃，官方可以支持一下对armv8-a向量寄存器的保存吗？

后面应该是会加的，这份向量寄存器可能是来源于 armv7 才没有加导致。

GuEe-GUI, 2023-07-17:
https://github.com/RT-Thread/rt-thread/pull/7815

closed 2023-07-29 by mysterywolf