RT-Thread/rt-thread #7719: stm32f103ve添加finsh组件报错bus fault: SCB_CFSR_BFSR:0x04 IMPRECISERR

jianglizi, 2023-06-25:
开发板野火指南者，
环境keil5 使用标准固件库，移植了rt-thread3.1.5，开启了RT_USING_CONSOLE，控制台输出正常，其后又添加了finsh组件，编译通过运行报错，下面为报错原内容，路过的大神帮帮，RT_USING_HEAP开启了size 2048，关闭Finsh组件又恢复正常

 \ | /
- RT -     Thread Operating System
 / | \     3.1.5 build Jun 25 2023
 2006 - 2020 Copyright by rt-thread team
psr: 0x21000000
r00: 0x20004414
r01: 0x20000408
r02: 0x00000000
r03: 0x2000436c
r04: 0x200003f4
r05: 0x00000000
r06: 0xdeadbeef
r07: 0xdeadbeef
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x00000000
 lr: 0x08002a45
 pc: 0x08002a6a
hard fault on thread: main

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    21  ready   0x00000040 0x00000400    06%   0x0000000a 000
tidle      7  ready   0x00000044 0x00000100    26%   0x00000020 000
main       2  ready   0x00000044 0x00000200    47%   0x00000013 000
bus fault:
SCB_CFSR_BFSR:0x04 IMPRECISERR

mysterywolf, 2023-06-25:
请到论坛提问 谢谢 issue用于报告bug

closed 2023-06-25 by mysterywolf

jianglizi, 2023-06-25:
>可以帮我看看嘛？ 我觉得它可能是个bug，因为我再三确认了移植步骤和添加的必要驱动函数的正确性，同时也检查了堆大小，还是发现不了错误

mysterywolf, 2023-06-25:
可以使用cmbacktrace软件包追踪一下函数的调用栈

jianglizi, 2023-06-25:
> 

好的俺去学习下