RT-Thread/rt-thread #7724: ulog支持打印调用所在的函数，文件名和行数

mysterywolf, 2023-06-26:
ulog支持打印调用所在的函数以及文件和行数.

如果一起都打印出来会导致日志内容过场，可以分别定义三个宏，用户可以自由决定开启打印 函数 文件名 行数

polarvid, 2023-07-23:
RT_ASSERT 中是否也可以添加文件、代码行的输出？

https://github.com/RT-Thread/rt-thread/blob/aee1bd532ed4b21e2eb2c5cfa32fd8af0b579bf3/include/rtthread.h#L715-L719

mysterywolf, 2023-07-23:
赞同

Guozhanxin, 2023-08-29:
> RT_ASSERT 中是否也可以添加文件名的输出？
> 
> https://github.com/RT-Thread/rt-thread/blob/aee1bd532ed4b21e2eb2c5cfa32fd8af0b579bf3/include/rtthread.h#L715-L719

好像文件名这个东西，不同的编译器行为不一样，我记得哪个编译器会把全路径打出来，就很烦人。。。

polarvid, 2023-09-06:
> > RT_ASSERT 中是否也可以添加文件名的输出？
> > https://github.com/RT-Thread/rt-thread/blob/aee1bd532ed4b21e2eb2c5cfa32fd8af0b579bf3/include/rtthread.h#L715-L719
> 
> 好像文件名这个东西，不同的编译器行为不一样，我记得哪个编译器会把全路径打出来，就很烦人。。。

VS Code 有路径和行数可以直接生成超链接跳转，还是很方便的。而且显示格式可以通过宏配置，这样有需要可以打开。特别是 assert。现在查 assert 还要全局搜索一下函数名，可能还有重名，查起来不是很方便……

armink, 2023-09-17:
文件名和行号信息对于断言是实用的，最好默认就有的。但对 ulog 而言，尤其是一些低级别日志，用处不见得多。以前 elog 我也加过，而且不同级别日志格式也是可配置的，但实用性较少，还会增加很多冗余 log ，后面 ulog 就暂时没做这块

a1012112796, 2023-09-27:
有额外flash 开销吧。。。

ymj0440, 2026-10-30:
/claim