RT-Thread/rt-thread #7725: RT-Thread 5.0.2 发布推进计划

mysterywolf, 2023-06-26:
RT-Thread v5.0.1 [已经发布](https://github.com/RT-Thread/rt-thread/releases/tag/v5.0.1) ，自RT-Thread 5.0.0发布之后，采用定期发布小版本迭代的方式，迭代的RT-Thread v5.1.0稳定版本。

本Issue将记录 RT-Thread v5.0.2发布推进计划：

### 文档
- [ ] 出一份社区 Code of Conduct，提交到github仓 @Cathy-lulu 

### 内核
- [ ] 将rtdebug.h归并到rtdbg.h里，并舍弃rtdebug.h，只保留rtdbg.h，否则两个debug头文件，已咨询过龙哥的建议 （rtdebug.h丢进legacy @仕成）
- [ ] 修复bsp/simulator vs2019编译下 atomic 两个volatile的问题
- [ ] 加入Device Driver v2.0重构，更好的支持到设备树；
- [ ] DFS v2中加入page cache，以及针对文件的mmap
- [ ] 针对MPU芯片，统一都使用SMP模型，对于单核，CPU数量=1；（影响到ARM Cortex-A，AArch64，RV64）
- [ ] exmaple文件夹里的内容已经过时或者非正式，考虑部分转移到utest，部分放在sample软件包里，将utest文件夹抽到根目录下？

### userapps
- [ ] 完成weston移植，加入SDL2到wayland client的适配

### ENV教程
- [ ] env-windows仓 QEMU需要升级 只升级64位
- [ ] Env教程和文档，
- [ ] Env教学视频重录 基本配置 怎么生成各种工程 唐同学的演讲 等
- [ ] formatting集成到env里 scons --formatting etc...
- [ ] 如何提交一个软件包

cgmjok, 2023-07-14:
您好 我是一个小白，我想请教下为什么官方提供的例子里面没有device.c文件，另外就是我在别的地方添加了device.c文件后对应的包是可以用KEIL编译过，但是在ENV下面还是会报错信息大概是:H:\AGVcode\rt-thread-master\components\drivers\core/device.c:462: first defined here
collect2.exe: error: ld returned 1 exit status
scons: *** [rt-thread.elf] Error 1
scons: building terminated because of errors.
如果有人花一点时间帮助我，我将不胜感激

mysterywolf, 2023-07-15:
请不要发布无关评论，请在论坛中提问 谢谢

BernardXiong, 2023-07-22:
后续是否可能考虑，

1. 在rtdef.h中屏蔽掉所有的任务，内核对象的定义；
2. 衍生一个rtobject.h的头文件，把相关定义放如这个头文件中；
3. rtobject.h的头文件谨慎包含，这样基本上不存在外部访问任务，内核对象的结构体，而只是使用指针；

BernardXiong, 2023-09-09:
> @mysterywolf "针对MPU芯片，统一都使用SMP模型，对于单核，CPU数量=1；" 这个指的是无论单核还是多核，都套用 CPU 模型吗？

好像还没有cpu模型？

mysterywolf, 2023-09-14:
- [ ] 修复--dist命令

closed 2023-10-23 by mysterywolf