RT-Thread/rt-thread #7764: [新手任务][BSP] 编译错误

supperthomas, 2023-07-01:
mm32  目录下面执行命令`scons`  编译错误
- [ ] mm32f103x
- [ ] mm32f327x
- [ ] mm32l07x
- [ ] mm32l3xx

renesas 目录下面执行命令`scons` 编译错误

- [ ] renesas/ra2l1-cpk
- [ ] renesas/ra4m2-eco

NXP 下面目录执行`scons` 编译错误

- [ ] imxrt/imxrt1021-nxp-evk
- [ ] imxrt/imxrt1170-nxp-evk/m7
- [ ] imx/imx6ull-smart
- [ ] 

Infine 
- [ ] Infineon/psoc6-pioneerkit_modus

其他

- [ ] w60x

supperthomas, 2023-07-13:
https://club.rt-thread.org/ask/article/8e019d0e64e1ab31.html

supperthomas, 2023-08-06:
参考action 的编译日志：
https://github.com/RT-Thread/rt-thread/actions/runs/5773375752

AndrewFrank-zxy, 2023-11-30:
有一个问题不知道在这个issue下讨论是否合适，我刚开始使用rt-thread，并使用[ubuntu系统构建开发环境](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/application-note/setup/qemu/ubuntu/an0005-qemu-ubuntu)，当在目录`rt-thread/bsp/qemu-vexpress-a9`中使用`scons --menuconfig`进行配置时，提示错误：
```bash
AttributeError: 'str' object has no attribute 'load_config':
  File "/home/XXXX/workspace/rt-thread/bsp/qemu-vexpress-a9/SConstruct", line 29:
    objs = PrepareBuilding(env, RTT_ROOT)
  File "/home/XXXX/workspace/rt-thread/bsp/qemu-vexpress-a9/../../tools/building.py", line 316:
    menuconfig(Rtt_Root)
  File "/home/XXXX/.local/lib/python3.8/site-packages/menuconfig.py", line 685:
    _conf_changed = _load_config()
  File "/home/XXXX/.local/lib/python3.8/site-packages/menuconfig.py", line 741:
    print(_kconf.load_config())
```
是我python的版本有问题吗？我应如何解决此问题？

supperthomas, 2023-12-04:
hpmicro/hpm6750evk
hpmicro/hpm6750evkmini
microchip/saml10
nuclei/gd32vf103_rvstar

```
scons: Entering directory `/home/runner/work/rt-thread/rt-thread/bsp/hpmicro/hpm6750evk'
scons: Reading SConscript files ...
AttributeError: module 'rtconfig' has no attribute 'DEVICE':
```

supperthomas, 2024-08-25:
❌ build nuvoton/numaker-hmi-ma35d1 failed.
❌ build nuvoton/numaker-m467hj failed.
❌ build nuvoton/numaker-m2354 failed.
❌ build nuvoton/ma35-rtp failed.
❌ build nuvoton/numaker-pfm-m487 failed.
❌ build nuvoton/nk-rtu980 failed.
❌ build nuvoton/nk-980iot failed.
❌ build nuvoton/numaker-m032ki failed.
❌ build nuvoton/nk-n9h30 failed.
❌ build nuvoton/numaker-iot-ma35d1 failed.
❌ build nuvoton/numaker-iot-m467 failed.
❌ build nuvoton/numaker-iot-m487 failed.
❌ build ft2004 failed.
❌ build yichip/yc3121-pos failed.
❌ build hc32l136 failed.
❌ build airm2m/air105 failed.
❌ build mm32l07x failed.
❌ build w60x failed.
❌ build apm32/apm32f030r8-miniboard failed.
❌ build apm32/apm32f051r8-evalboard failed.
❌ build nxp/lpc/lpc54114-lite failed.
❌ build nxp/lpc/lpc5410x failed.
❌ build nxp/lpc/lpc54608-LPCXpresso failed.
❌ build nxp/imx/imxrt/imxrt1170-nxp-evk/m7 failed.
❌ build nxp/imx/imx6ull-smart failed.
❌ build raspberry-pi/raspi2 failed.
❌ build essemi/es32f0654 failed.
❌ build essemi/es32f365x failed.
❌ build ti/c28x/tms320f28379d failed.
❌ build wch/arm/ch579m failed.
❌ build microchip/saml10 failed.
❌ build ht32/ht32f52352 failed.
❌ build ht32/ht32f12366 failed.
❌ build fm33lc026 failed.
❌ build amebaz failed.

https://github.com/supperthomas/rt-thread/actions/runs/10545116916

supperthomas, 2024-08-26:
顺便验证下 scons --dist是否正常