RT-Thread/rt-thread #7777: [LPC4088] IAR生成工程多个配置参数_DLIB_THREAD_SUPPORT

meng-plus, 2023-07-04:
scons --target=iar 生成的工程会多增加一个_DLIB_THREAD_SUPPORT
导致编译提示
> 	Fatal Error[Pe035]: #error directive: "Option has been removed"	D:\Program Files\IAR Systems\Embedded Workbench 9.1\arm\inc\c\DLib_Defaults.h	596	

需要手动移除 options->C/C++ Compiler->Preocessor->Defined sysmbols 配置的_DLIB_THREAD_SUPPORT

meng-plus, 2023-07-04:
怀疑问题点在这
https://github.com/RT-Thread/rt-thread/blob/0000000000000000000000000000000000000000:/components/libc/compilers/dlib/SConscript#L16

mysterywolf, 2023-07-04:
你用的IAR版本是多少？ 麻烦请给我往我邮件随便发些什么 meco@rt-thread.org

meng-plus, 2023-07-04:
> 你用的IAR版本是多少？ 麻烦请给我往我邮件随便发些什么 meco@rt-thread.org

我用的 9.30 我iar工程路径没有正确配置 他无法识别到版本号

mysterywolf, 2023-08-17:
我看了一下 用menuconfig -s 设置一下IAR所在目录就可以了

[image]

closed 2023-08-17 by mysterywolf