RT-Thread/rt-thread #7820: RISC-V 移植中没有初始化线程的thread-pointer

helloeagleyang, 2023-07-14:
RT-Thread 移植的libcpu/risc-v/common/cpuport.c中没有初始化任务栈的thread-pointer，在线程中调用到依赖thread-pointer的gcc库函数后，可能会导致异常（因为tp默认被初始化0xdeadbeef，这个地址在RISC-V 芯片中大部分是非法地址）

BernardXiong, 2023-07-22:
请问你使用的BSP是哪个，调用什么样的gcc（or libc？）函数会出错？

helloeagleyang, 2023-07-27:
> 请问你使用的BSP是哪个，调用什么样的gcc（or libc？）函数会出错？

我反汇编了基于rt-thread v5.0.1的先楫的rw007_wifi代码，发现了对tp的多处引用，包括网络部分,如下面的代码所示，如果不正确的初始化tp，代码很容易崩溃

`        dns_setserver(dns_num, &addr);
 1080f8c:       006c                    addi    a1,sp,12
 1080f8e:       8522                    mv      a0,s0
 1080f90:       7ef8d097                auipc   ra,0x7ef8d
 1080f94:       342080e7                jalr    834(ra) # 8000e2d2 <dns_setserver>
}
 1080f98:       40f2                    lw      ra,28(sp)
 1080f9a:       4462                    lw      s0,24(sp)
 1080f9c:       6105                    addi    sp,sp,32
 1080f9e:       8082                    ret
 1080fa0:       2820                    .2byte  0x2820
 1080fa2:       6544                    .2byte  0x6544
 1080fa4:       6166                    .2byte  0x6166
 1080fa6:       6c75                    lui     s8,0x1d
 1080fa8:       2974                    .2byte  0x2974
 1080faa:       0000                    unimp
 1080fac:       6e64                    .2byte  0x6e64
 1080fae:       65732073                csrs    0x657,t1
 1080fb2:       7672                    .2byte  0x7672
 1080fb4:       7265                    lui     tp,0xffff9
 1080fb6:       2320                    .2byte  0x2320
 1080fb8:       6425                    lui     s0,0x9
 1080fba:       203a                    .2byte  0x203a
 1080fbc:       7325                    lui     t1,0xfffe9
 1080fbe:       000a                    c.slli  zero,0x2
 1080fc0:       656e                    .2byte  0x656e
 1080fc2:       7774                    .2byte  0x7774
 1080fc4:       206b726f                jal     tp,11381ca <__rt_rvstack+0x594aa>
 1080fc8:       6e69                    lui     t3,0x1a`

MuChenger, 2026-03-11:
tp在没使用__thread等定义时编译器是不使用的，上述的tp应该是返汇编误翻译的