RT-Thread/rt-thread #7824: 新唐的bsp包，都不能正常运行

linhuikui, 2023-07-17:
新唐(Nuvoton)的NUC980的bsp包，不能正常运行，里面所有的外设，没有对引脚进行功能设置，m03包也有同样问题。

supperthomas, 2023-07-17:
欢迎pr

supperthomas, 2023-07-19:
[wosayttn](https://github.com/wosayttn)
@wosayttn

wosayttn, 2023-07-20:
> 新唐(Nuvoton)的NUC980的bsp包，不能正常运行，里面所有的外设，没有对引脚进行功能设置，m03包也有同样问题。

Hi,

All multi-function pin settings are collected in board folder in nuvoton porting.
You should modify or implement pin function initialization function by your self. 

For example, see our nk-980iot and numaker-m032ki development kit.

https://github.com/RT-Thread/rt-thread/blob/master/bsp/nuvoton/nk-980iot/board/nu_pin_init.c
- All multi-function pin settings are in nu_pin_init function.

https://github.com/RT-Thread/rt-thread/blob/master/bsp/nuvoton/numaker-m032ki/board/NuPinConfig/nutool_pincfg.c
- All multi-function pin settings are in nutool_pincfg_init function.

closed 2023-07-29 by mysterywolf