RT-Thread/rt-thread #7860: rt_timer_check函数全程关中断有可能影响中断函数响应时间，其函数的关中断时间是否能优化？

BreederBai, 2023-07-24:
芯片：HC32F4A0
问题描述：定时器捕获时发现中断响应有时会延时20多us，单步调试发现是由于rt_timer_check函数长时间关中断造成的。
0通道：捕获信号；1通道：测试用的IO，在捕获中断（上升沿）函数中反转电平；2通道：rt_timer_check函数，在关中断和开中断时反转电平
[image]

BernardXiong, 2023-07-25:
所以这个也会和用户的timer定时器会执行什么超时函数相关

BreederBai, 2023-07-25:
大部分执行的是_thread_timeout函数，调用rt_thread_mdelay函数，超时后是会执行_thread_timeout吗？

BreederBai, 2023-08-07:
> 所以这个和用户的定时器计时器会执行什么相关的超时函数

熊大，这个问题即使和用户定时器执行的超时函数有关，似乎也不应该在关中断里执行用户的超时程序

Blues-Jiang, 2023-08-07:
我也查到t->timeout_func(t->parameter);这个函数了，但是我最小系统，这里是哪个定时器的影响呢，我本身没开任何定时器啊。

Blues-Jiang, 2023-08-07:
主定时器的话超时函数就是_thread_timeout，应该就是t->timeout_func(t->parameter);对应的函数，这个会运行8.6us大概（加上我控制io的时间），我是GD32, 然后rt_schedule会运行4us左右，
rt_thread_sleep里也会运行将近9us，里面耗时的就是    err = rt_thread_suspend_with_flag(thread, RT_INTERRUPTIBLE);//4.6us
        
rt_timer_control(&(thread->thread_timer), RT_TIMER_CTRL_SET_TIME, &tick);
rt_timer_start(&(thread->thread_timer));//4.7us

再往下我也找不到具体源头了，就我现在这个代码 加上io控制，已经几乎30+us不开中断了，每个时间段都有10us左右，打5折总delay也得15us

BreederBai, 2023-08-07:
> 我也查到t->timeout_func(t->parameter);这个函数了，但是我最小系统，这里是哪个定时器的影响呢，我本身没开任何定时器啊。

有可能是线程的定时器吧

Blues-Jiang, 2023-08-07:
> > 我也查到t->timeout_func(t->parameter);这个函数了，但是我最小系统，这里是哪个定时器的影响呢，我本身没开任何定时器啊。
> 
> 有可能是线程的定时器吧

system timer thread 就是默认开的软定时器，_timer_thread具体是这个线程

Blues-Jiang, 2023-08-07:
> > 我也查到t->timeout_func(t->parameter);这个函数了，但是我最小系统，这里是哪个定时器的影响呢，我本身没开任何定时器啊。
> 
> 有可能是线程的定时器吧

应该是了，线程对应的定时器。这个东西是挺耗时间的