RT-Thread/rt-thread #7862: bsp\wch\risc-v\ch32v307v-r1编译出错，直接拉最新代码未作任何修改。

kylongmu, 2023-07-25:
`D:\Develop\Git_Project\rt-thread\bsp\wch\risc-v\ch32v307v-r1
> scons --exec-path=D:\Develop\sdk-toolchain-RISC-V-GCC-WCH-1.0.0\bin
scons: Reading SConscript files ...
Newlib version: 3.0.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\main.o
CC build\board\board.o
CC build\board\software_irq.o
CC build\kernel\components\drivers\core\device.o
CC build\kernel\components\drivers\ipc\completion.o
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c: In function 'rt_completion_wait':
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c:61:5: warning: implicit declaration of function 'RT_DEBUG_SCHEDULER_AVAILABLE' [-Wimplicit-function-declaration]
     RT_DEBUG_SCHEDULER_AVAILABLE(timeout != 0);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c:89:13: error: 'RT_DEBUG_NOT_IN_INTERRUPT' undeclared (first use in this function)
             RT_DEBUG_NOT_IN_INTERRUPT;
             ^~~~~~~~~~~~~~~~~~~~~~~~~
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c:89:13: note: each undeclared identifier is reported only once for each function it appears in
scons: *** [build\kernel\components\drivers\ipc\completion.o] Error 1
scons: building terminated because of errors.`

mysterywolf, 2023-07-25:
尝试更新一下代码，主仓代码是有定义RT_DEBUG_NOT_IN_INTERRUPT 的

kylongmu, 2023-07-26:
> 尝试更新一下代码，主仓代码是有定义RT_DEBUG_NOT_IN_INTERRUPT 的

更新代码后，menuconfig再执行一下可以编译通过。
然后合并进自己的工程，但是以前编译正常的工程，现在无法链接
`LINK rtthread.elf                                                                                                                                                                        
d:/develop/sdk-toolchain-risc-v-gcc-wch-1.0.0/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: rtthread.elf section `.text' will not fit in region `FLASH' 
d:/develop/sdk-toolchain-risc-v-gcc-wch-1.0.0/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: region `FLASH' overflowed by 11832 bytes                    
collect2.exe: error: ld returned 1 exit status                                                                                                                                           
scons: *** [rtthread.elf] Error 1                                                                                                                                                        
scons: building terminated because of errors.                                                                                                                                            `

kylongmu, 2023-07-26:
我看了下我原先工程里面Enable debugging features这里是没有选中的，编译链接都正常。
更新git源编译例子工程，里面的Enable debugging features是选中的，可以编译通过，但是去掉这个选中后，就会出以下问题：
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c:89:13: error: 'RT_DEBUG_NOT_IN_INTERRUPT' undeclared (first use in this function)
RT_DEBUG_NOT_IN_INTERRUPT;

我的工程合并git后，没选中Enable debugging features一样报错，选中后就出现上面的链接失败：FLASH' overflowed

aozima, 2023-07-26:
应该是新版本代码体积变大了，可以先改链接脚本，把FLASH加大些，让链接通过后，再根据map文件，看看哪些可以裁剪优化。

kylongmu, 2023-07-26:
> 应该是新版本代码体积变大了，可以先改链接脚本，把FLASH加大些，让链接通过后，再根据map文件，看看哪些可以裁剪优化。

这是我没有合并新代码的编译尺寸：
   text    data     bss     dec     hex filename
 208320    2976    7816  219112   357e8 rtthread.elf

307有224K的FLASH，即229376，还有10264的剩余，合并新代码后必须选中Enable debugging features才能编译通过，结果反而溢出了11832，也就是说这个选项一下就要增加22KB左右空间占用。我为什么不能不用这个选项？而要去费劲的查那个功能不需要用来优化呢？

kylongmu, 2023-07-26:
现在明显就是不选中Enable debugging features，会产生这个编译错误：
D:\Develop\Git_Project\rt-thread\components\drivers\ipc\completion.c:89:13: error: 'RT_DEBUG_NOT_IN_INTERRUPT' undeclared (first use in this function)
RT_DEBUG_NOT_IN_INTERRUPT;

mysterywolf, 2023-07-27:
我理解了，已经在此处修复 https://github.com/RT-Thread/rt-thread/pull/7875 感谢反馈

closed 2023-07-27 by mysterywolf