RT-Thread/rt-thread #7872: workqueue编译报错

heyuanjie87, 2023-07-26:
要是把#include <ipc/workqueue.h>写在c文件的第一行就会报如下编译错误：

In file included from G:\adas\software\kernel\components\legacy/rtlegacy.h:34,
                 from G:\adas\software\kernel\include/rtthread.h:33,
                 from G:\adas\software\kernel\components\drivers\include/ipc/workqueue.h:14,
                 from applications\main.c:10:
G:\adas\software\kernel\components\legacy/ipc/workqueue_legacy.h:18:20: error: field 'work' has incomplete type
   18 |     struct rt_work work;
      |                    ^~~~

----------------
编译器rt-smart gcc

Guozhanxin, 2023-07-27:
你这个很明显就是找不到结构体 struct rt_work 的定义啊，是在那个bsp上编译的啊

mysterywolf, 2023-07-27:
是要兼容什么东西嘛? 为什么要把legacy code打开

heyuanjie87, 2023-07-27:
> 你这个很明显就是找不到结构体 struct rt_work 的定义啊，是在那个bsp上编译的啊

按道理不应该找不到才正常,和bsp无关，我在项目上报错后在main.c包含测试也报错

heyuanjie87, 2023-07-27:
> 是要兼容什么东西嘛? 为什么要把legacy code打开

项目代码里其他同事打开的兼容，workqueue可以不放legacy里吗，怀疑是头文件循环包含把gcc搞懵逼了

mysterywolf, 2023-07-28:
可以的 我觉得没啥兼容的必要了

mysterywolf, 2023-07-28:
我提交了一个PR修复这个问题 https://github.com/RT-Thread/rt-thread/pull/7883  可以再验证一下

closed 2023-07-29 by mysterywolf