RT-Thread/rt-thread #7873: [warning] Kconfig中有一些warning 如果觉得有问题的可以尝试PR

supperthomas, 2023-07-26:
找到任意bsp
执行命令`scons --pyconfig-silent`

```
Kconfig:5: warning: BSP_ROOT has 'option env="BSP_DIR"', but the environment variable BSP_DIR is not set
Kconfig:5: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename BSP_DIR to BSP_ROOT (so that the symbol name matches the environment variable name).
Kconfig:10: warning: RTT_ROOT has 'option env="RTT_DIR"', but the environment variable RTT_DIR is not set
Kconfig:10: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename RTT_DIR to RTT_ROOT (so that the symbol name matches the environment variable name).
Kconfig:15: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename PKGS_DIR to PKGS_ROOT (so that the symbol name matches the environment variable name).
warning: FDT_USING_DEBUG (defined at ../../components/drivers/Kconfig:228) has leading or trailing whitespace in its prompt
warning: RT_VBUS_USING_TESTS (defined at ../../components/vbus/Kconfig:20) has leading or trailing whitespace in its prompt
warning: RT_VBUS_GUEST_VIRQ (defined at ../../components/vbus/Kconfig:34) defined with more than one help text -- only the last one will be used
/root/.env/packages/packages/iot/lorawan_driver/Kconfig:33: warning: style: quotes recommended around 'ClassA' in 'bool ClassA'
/root/.env/packages/packages/iot/lorawan_driver/Kconfig:35: warning: style: quotes recommended around 'ClassC' in 'bool ClassC'
/root/.env/packages/packages/iot/lorawan_driver/Kconfig:44: warning: style: quotes recommended around 'OTAA' in 'bool OTAA'
/root/.env/packages/packages/iot/lorawan_driver/Kconfig:47: warning: style: quotes recommended around 'ABP' in 'bool ABP'
warning: SALOF_LOG_LEVEL (defined at /root/.env/packages/packages/iot/kawaii-mqtt/Kconfig:76) has leading or trailing whitespace in its prompt
/root/.env/packages/packages/multimedia/gui_engine/Kconfig:32: warning: style: quotes recommended around '32' in 'bool 32'
/root/.env/packages/packages/multimedia/gui_engine/Kconfig:35: warning: style: quotes recommended around '24' in 'bool 24'
warning: RTT_CMD_LOOP_1_DELAY_TIME (defined at /root/.env/packages/packages/tools/rtt_auto_exe_cmd/Kconfig:47) has leading or trailing whitespace in its prompt
warning: CW2015_USE_AUTO_UPDATE (defined at /root/.env/packages/packages/peripherals/sensors/cw2015/Kconfig:17) has leading or trailing whitespace in its prompt
warning: PKG_USING_DRV_INFRARED (defined at /root/.env/packages/packages/peripherals/infrared/Kconfig:24) has leading or trailing whitespace in its prompt
warning: PKG_USING_DRV_MULTI_INFRARED (defined at /root/.env/packages/packages/peripherals/multi_infrared/Kconfig:19) has leading or trailing whitespace in its prompt
warning: VDEVICE_USING_PIN (defined at /root/.env/packages/packages/peripherals/vdevice/Kconfig:13) has leading or trailing whitespace in its prompt
warning: VDEVICE_USING_LCD (defined at /root/.env/packages/packages/peripherals/vdevice/Kconfig:24) has leading or trailing whitespace in its prompt
warning: VDEVICE_USING_TOUCH (defined at /root/.env/packages/packages/peripherals/vdevice/Kconfig:34) has leading or trailing whitespace in its prompt
warning: <choice> (defined at drivers/Kconfig:17) has leading or trailing whitespace in its prompt
warning: style: quotes recommended around default value for string symbol RT_LWIP_IPADDR (defined at ../../components/net/lwip/Kconfig:93)
warning: style: quotes recommended around default value for string symbol RT_LWIP_GWADDR (defined at ../../components/net/lwip/Kconfig:97)
warning: style: quotes recommended around default value for string symbol RT_LWIP_MSKADDR (defined at ../../components/net/lwip/Kconfig:101)
warning: style: quotes recommended around default value for string symbol DHCPD_SERVER_IP (defined at ../../components/net/lwip/Kconfig:262)
warning: style: quotes recommended around default value for string symbol DHCP_DNS_SERVER_IP (defined at ../../components/net/lwip/Kconfig:274)
warning: style: quotes recommended around default value for string symbol WIZ_IPADDR (defined at /root/.env/packages/packages/iot/wiznet/Kconfig:58)
warning: style: quotes recommended around default value for string symbol WIZ_GWADDR (defined at /root/.env/packages/packages/iot/wiznet/Kconfig:62)
warning: style: quotes recommended around default value for string symbol WIZ_MSKADDR (defined at /root/.env/packages/packages/iot/wiznet/Kconfig:66)
warning: style: quotes recommended around default value for string symbol PKG_USING_ALI_IOTKIT_PRODUCT_KEY (defined at /root/.env/packages/packages/iot/iot_cloud/ali-iotkit/Kconfig:17)
warning: style: quotes recommended around default value for string symbol PKG_USING_ALI_IOTKIT_PRODUCT_SECRET (defined at /root/.env/packages/packages/iot/iot_cloud/ali-iotkit/Kconfig:21)
warning: style: quotes recommended around default value for string symbol PKG_USING_ALI_IOTKIT_DEVICE_SECRET (defined at /root/.env/packages/packages/iot/iot_cloud/ali-iotkit/Kconfig:29)
warning: style: quotes recommended around default value for string symbol PKG_USING_TENCENT_IOT_EXPLORER_DEVICE_NAME (defined at /root/.env/packages/packages/iot/iot_cloud/tencent-iot-sdk/Kconfig:20)
warning: style: quotes recommended around default value for string symbol LORAWAN_ED_STACK_PHY_PARAMETER_CHANNEL_MASK1 (defined at /root/.env/packages/packages/iot/lorawan_ed_stack/Kconfig:243)
warning: style: quotes recommended around default value for string symbol LORAWAN_ED_STACK_PHY_PARAMETER_CHANNEL_MASK2 (defined at /root/.env/packages/packages/iot/lorawan_ed_stack/Kconfig:246)
warning: style: quotes recommended around default value for string symbol LORAWAN_ED_STACK_PHY_PARAMETER_CHANNEL_MASK3 (defined at /root/.env/packages/packages/iot/lorawan_ed_stack/Kconfig:249)
warning: style: quotes recommended around default value for string symbol LORAWAN_ED_STACK_PHY_PARAMETER_CHANNEL_MASK4 (defined at /root/.env/packages/packages/iot/lorawan_ed_stack/Kconfig:252)
warning: style: quotes recommended around default value for string symbol LORAWAN_ED_STACK_PHY_PARAMETER_CHANNEL_MASK5 (defined at /root/.env/packages/packages/iot/lorawan_ed_stack/Kconfig:255)
warning: style: quotes recommended around default value for string symbol MP3_SOUND_DEVICE_NAME (defined at /root/.env/packages/packages/multimedia/mp3player/Kconfig:16)
warning: style: quotes recommended around default value for string symbol LORA_RADIO0_DEVICE_NAME (defined at /root/.env/packages/packages/peripherals/lora_radio_driver/Kconfig:33)
warning: style: quotes recommended around default value for string symbol LORA_RADIO0_SPI_BUS_NAME (defined at /root/.env/packages/packages/peripherals/lora_radio_driver/Kconfig:37)
warning: style: quotes recommended around default value for string symbol LORA_GW_DRIVER_SPI_BUS_NAME (defined at /root/.env/packages/packages/peripherals/lora_gw_driver_lib/Kconfig:27)
warning: style: quotes recommended around default value for string symbol LORA_GW_DRIVER_I2C_DEVICE_NAME (defined at /root/.env/packages/packages/peripherals/lora_gw_driver_lib/Kconfig:33)
warning: style: quotes recommended around default value for string symbol LORA_MODEM_UART_DEVICE (defined at /root/.env/packages/packages/peripherals/lora_modem_driver/Kconfig:13)
warning: the int symbol RTDUINO_THREAD_PRIO (defined at /root/.env/packages/packages/arduino/RTduino/Kconfig:24) has a non-int default RT_THREAD_PRIORITY_MAX-2 (undefined)
warning: the choice symbol RT_USING_USERHEAP (defined at ../../src/Kconfig:343) is selected by the following symbols, but select/imply has no effect on choice symbols
 - PKG_USING_TLSF (defined at /root/.env/packages/packages/system/tlsf/Kconfig:3)
warning: the choice symbol RT_USING_SERIAL_V1 (defined at ../../components/drivers/Kconfig:37) is selected by the following symbols, but select/imply has no effect on choice symbols
 - BSP_USING_UART (defined at drivers/Kconfig:59)
warning: the default selection ClassA (undefined) of <choice> (defined at /root/.env/packages/packages/iot/lorawan_driver/Kconfig:27) is not contained in the choice
warning: the default selection ABP (undefined) of <choice> (defined at /root/.env/packages/packages/iot/lorawan_driver/Kconfig:38) is not contained in the choice
[... cut ...]
warning: the default selection ABUP_NETWORKING_MODE (undefined) of <choice> (defined at /root/.env/packages/packages/iot/abup_fota/Kconfig:14) is not contained in the choice
warning: the default selection ABUP_ALGORITHM (undefined) of <choice> (defined at /root/.env/packages/packages/iot/abup_fota/Kconfig:39) is not contained in the choice
warning: the default selection RT_LINK_USING_UART (undefined) of <choice> (defined at /root/.env/packages/packages/iot/rt-link_hw/Kconfig:8) is not contained in the choice

```

BernardXiong, 2023-07-29:
确实，这个警告有些多。用pyconfig和使用kconfig-frontend的mconf还不一样。

supperthomas, 2024-01-04:
```
> scons --pyconfig-silent
scons: Reading SConscript files ...
Kconfig:5: warning: BSP_ROOT has 'option env="BSP_DIR"', but the environment variable BSP_DIR is not set
Kconfig:5: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename BSP_DIR to BSP_ROOT (so that the symbol name matches the environment variable name).
Kconfig:10: warning: RTT_ROOT has 'option env="RTT_DIR"', but the environment variable RTT_DIR is not set
Kconfig:10: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename RTT_DIR to RTT_ROOT (so that the symbol name matches the environment variable name).
Kconfig:15: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename PKGS_DIR to PKGS_ROOT (so that the symbol name matches the environment variable name).
Loaded configuration '.config'
Configuration saved to '.config'
```

closed 2024-01-04 by supperthomas

mysterywolf, 2024-01-04:
option env="BSP_DIR" 有机会消掉吗？