RT-Thread/rt-thread #7912: Raspberry Pi 4 64 bit sample code will not build without modifications

DerrickLau, 2023-08-04:
I grabbed the latest version of code from the master branch.  I installed RT-Thread toolset, and I modified the rtconfig.py file in the Raspberry Pi 64 by moving RTT_ROOT assignment to below the if/else statement that assigns its value (essentially hardcoding it).  I then typed SCONS into the console emulator and produced several errors.  I followed the instructions of the errors and modified the source code.

I then modified the following source:
C:\Derrick\GitHub\rt-thread-master-vanilla\rt-thread-master\components\libc\compilers\common\ctime.c
Changed line 700 #include to full path to resource_id.h
Changed line 906 by commenting out call to resource_id_get.
Changed line 944 by commenting out call to resource_id_put.
Ran SCONS and now it built.

My issue is why is this happening in the main/master branch?  That is where all the good code is merged in and should be tested regularly, so that when casual users like me download rt-thread, we always go to the main branch and expect it to at least build without issues.

BernardXiong, 2023-08-05:
Thank you for your feedback. Yes, it should been build successfully without modification.

There is a CI build for aarch64 bsps:

https://github.com/RT-Thread/rt-thread/actions/runs/5768201405/job/15638906829?pr=7917

Could you please check this result? If there are some issues, welcome to submit a PR (for CI build or BSP code etc). Thank you.

DerrickLau, 2023-08-06:
I believe you.  However, I will point out you did not build from the master branch.  Which branch was that built from?

DerrickLau, 2023-08-06:
I checked that result in another discussion and yes, the issue is on my end.  Your toolchain seems dependent on Ubuntu UNIX but I am building on a Windows laptop and downloaded the ARM Windows toolchain.
So don't let me check in any code.  ;)

closed 2023-08-06 by DerrickLau