RT-Thread/rt-thread #7913: N32的bsp包PIN设备的函数有部分编译警告

Xyasen, 2023-08-04:
在 [https://github.com/RT-Thread/rt-thread/blob/master/bsp/n32/libraries/n32_drivers/drv_gpio.c](url)中
```
const static struct rt_pin_ops _n32_pin_ops =
{
    n32_pin_mode,
    n32_pin_write,
    n32_pin_read,
    n32_pin_attach_irq,
    n32_pin_dettach_irq,
    n32_pin_irq_enable,
};
```

- n32_pin_mode
- n32_pin_write
- n32_pin_read
- n32_pin_attach_irq   
- n32_pin_irq_enable

编译会产生“warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]”

看了一下，指向的这个几个函数的参数变量类型都存在和 struct rt_pin_ops 定义不一样的地方
示例：
```
void (*pin_mode)(struct rt_device *device, rt_base_t pin, rt_uint8_t mode)

static void n32_pin_mode(rt_device_t dev, rt_base_t pin, rt_base_t mode)
```
mode应该为rt_uint8_t 却写为 rt_base_t 

不知道这个是不是失误，还是有什么特殊原因要这样写？

aozima, 2023-08-04:
可能是更新后，bsp里面没同步更新造成，可以git看下历史记录确认下，然后把bsp里面像其它bsp一样，做一下同步。

Xyasen, 2023-08-04:
> 可能是更新后，bsp里面没同步更新造成，可以git看下历史记录确认下，然后把bsp里面像其它bsp一样，做一下同步。

在drv_spi.c的
```
static struct rt_spi_ops spi_ops =
{
    configure,
    xfer
};
```
xfer也是如此，我看了一眼，好像没啥特殊原因，我自己改了先，后续版本应该会更新这个吧

aozima, 2023-08-04:
你改了可以直接发个PR 👍

Xyasen, 2023-08-04:
> 

> 你改了可以直接发个PR 👍

好的，谢谢你的板子。

closed 2023-08-04 by Xyasen