RT-Thread/rt-thread #7918: after scons --dist  , menuconfig will error

supperthomas, 2023-08-05:
after scons --dist  in 

`stm32/stm32f407-atk-explorer/ `

in dist project
execute `menuconfig`
will error like this:

```
KconfigError: rt-thread/libcpu/Kconfig:241: 'rt-thread/libcpu/aarch64/Kconfig' not found (in 'source "$RTT_DIR/libcpu/aarch64/Kconfig"').
```

```
Kconfig:5: warning: BSP_ROOT has 'option env="BSP_DIR"', but the environment variable BSP_DIR is not set
Kconfig:5: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename BSP_DIR to BSP_ROOT (so that the symbol name matches the environment variable name).
Kconfig:10: warning: RTT_ROOT has 'option env="RTT_DIR"', but the environment variable RTT_DIR is not set
Kconfig:10: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename RTT_DIR to RTT_ROOT (so that the symbol name matches the environment variable name).
Kconfig:15: warning: Kconfiglib expands environment variables in strings directly, meaning you do not need 'option env=...' "bounce" symbols. For compatibility with the C tools, rename PKGS_DIR to PKGS_ROOT (so that the symbol name matches the environment variable name).
KconfigError: rt-thread/libcpu/Kconfig:241: 'rt-thread/libcpu/aarch64/Kconfig' not found (in 'source "$RTT_DIR/libcpu/aarch64/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.:
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/SConstruct", line 52:
    objs = PrepareBuilding(env, RTT_ROOT, has_libcpu=False)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/building.py", line 352:
    guiconfig_silent(Rtt_Root)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/menuconfig.py", line 325:
    defconfig.main()
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/defconfig.py", line 37:
    kconf = kconfiglib.Kconfig(args.kconfig)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/kconfiglib.py", line 1030:
    self._parse_block(None, self.top_node, self.top_node)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/kconfiglib.py", line 2877:
    prev = self._parse_block(None, parent, prev)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/kconfiglib.py", line 2877:
    prev = self._parse_block(None, parent, prev)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/kconfiglib.py", line 2[89](https://github.com/lt6210925/rt-thread/actions/runs/5769480305/job/15641554912#step:7:90)6:
    self._parse_block(_T_ENDIF, node, node)
  File "/home/runner/work/rt-thread/rt-thread/bsp/stm32/stm32f407-atk-explorer/dist/project/rt-thread/tools/kconfiglib.py", line 2865:
    raise KconfigError(
```

BernardXiong, 2023-08-05:
It's best to add some scons checking in CI.

supperthomas, 2023-08-05:
action_tools.yml need add
```
        scons --dist-ide -C $TEST_BSP_ROOT
        ls $TEST_BSP_ROOT
        ls $TEST_BSP_ROOT/dist
        scons --pyconfig-silent -C $TEST_BSP_ROOT/dist/project
        scons --pyconfig-silent -C $TEST_BSP_ROOT/rt-studio-project
        scons -C $TEST_BSP_ROOT/dist/project
        scons -C $TEST_BSP_ROOT/rt-studio-project
```

mysterywolf, 2023-08-05:
最新的代码还有这个问题吗

supperthomas, 2023-08-06:
> 最新的代码还有这个问题吗，之前有一个PR尝试修复这个问题了
> https://github.com/RT-Thread/rt-thread/blob/master/libcpu/Kconfig#L241

你试下看看，可能没问题了，没问题就关了吧

supperthomas, 2023-08-06:
https://github.com/RT-Thread/rt-thread/actions/runs/5775067712/job/15652664439

supperthomas, 2023-08-06:
看起来还是有问题

supperthomas, 2023-08-07:
```
if ARCH_ARMV8 && ARCH_CPU_64BIT
    source "$RTT_DIR/libcpu/aarch64/Kconfig"
endif
```

这个只能保证menuconfig的时候不显示，但是如果找不到这个Kconfig，menuconfig还是会报错，所以`scons --dist` 会报错。

fengyang-iot, 2023-09-14:
如果要修复这个bug,得修改env工具里的scons --dist 命令处理过程吧，不可能每次自己动手修改文件

supperthomas, 2023-09-14:
> 如果要修复这个bug,得修改env工具里的scons --dist 命令处理过程吧，不可能每次自己动手修改文件

这个目前代码需要注释掉这段。
其实要是能完善下Kconfig，能够支持osource，说不定也能解决这个问题。

closed 2023-09-27 by zhkag