RT-Thread/rt-thread #7923: 最新master编译qemu-virt64-aarch64失败

FragrantRye, 2023-08-06:
```bash
user@DESKTOP-USER:~/source/rt-thread/bsp/qemu-virt64-aarch64$ git describe
v2.0.0_beta-11974-g8dfb3d14a

user@DESKTOP-USER:~/source/rt-thread/bsp/qemu-virt64-aarch64$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

user@DESKTOP-USER:~/source/rt-thread/bsp/qemu-virt64-aarch64$ $RTT_EXEC_PATH/aarch64-none-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/user/arm-gnu-toolchain-12.3.rel1-x86_64-aarch64-none-elf/bin//aarch64-none-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/arm-gnu-toolchain-12.3.rel1-x86_64-aarch64-none-elf/bin/../libexec/gcc/aarch64-none-elf/12.3.1/lto-wrapper
Target: aarch64-none-elf
Configured with: /data/jenkins/workspace/GNU-toolchain/arm-12/src/gcc/configure --target=aarch64-none-elf --prefix=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/install --with-gmp=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/host-tools --with-mpfr=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/host-tools --with-mpc=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/host-tools --with-isl=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-gnu-as --with-gnu-ld --with-sysroot=/data/jenkins/workspace/GNU-toolchain/arm-12/build-aarch64-none-elf/install/aarch64-none-elf --with-pkgversion='Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.3.1 20230626 (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35))

user@DESKTOP-USER:~/source/rt-thread/bsp/qemu-virt64-aarch64$ scons
scons: Reading SConscript files ...
Newlib version: 4.3.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/kernel/components/libc/posix/io/stdio/libc.o
/home/user/source/rt-thread/components/libc/posix/io/stdio/libc.c: In function 'libc_stdio_set_console':
/home/user/source/rt-thread/components/libc/posix/io/stdio/libc.c:124:22: error: 'struct _reent' has no member named '__sdidinit'
  124 |         _GLOBAL_REENT->__sdidinit = 1;
      |                      ^~
scons: *** [build/kernel/components/libc/posix/io/stdio/libc.o] Error 1
scons: building terminated because of errors.
```

BernardXiong, 2023-08-07:
估计哪里不太对，可以参考ci build里的情况来对比下环境

supperthomas, 2023-08-07:
> 估计哪里不太对，可以参考ci build里的情况来对比下环境

https://github.com/RT-Thread/rt-thread/discussions/7916
怎么跟这里报的问题一样的？
是不是rtt里面有用到linux里面的头文件，所以linux可以编译通过？windows编译不过？

supperthomas, 2023-08-07:
@mysterywolf struct _reent  这个定义在哪里定义的？

mysterywolf, 2023-08-07:
[image]
 

这句话一直就没有动过

supperthomas, 2023-08-07:
arm-gnu-toolchain-12.3.rel1-x86_64-aarch64-none-elf
我感觉可能toolchain变了。

mysterywolf, 2023-08-07:
有没有什么标志宏可以区分一下？

FragrantRye, 2023-08-14:
可以理解为当前rtt不支持12.3.1的toolchain吗？

supperthomas, 2023-08-14:
> ```shell
> __sdidinit 
> ```

components/libc/posix/io/stdio/libc.c 这里用到的toolchain里面的一些变量__sdidinit ， 这个在之前的toolchain里面有定义，最新的可能去掉了。所以如果用到这个相关的代码并且用12.3.1 的toolchain就暂时不支持。如果知道如何修改，可以尝试PR。用toolchain宏版本号隔开。

mysterywolf, 2023-08-16:
https://github.com/mirror/newlib-cygwin/commit/44b60f0c4ba597c55dcac4f2d3119a3055c80ba1 newlib在这个commit里删除掉了

mysterywolf, 2023-08-17:
https://github.com/RT-Thread/rt-thread/pull/7930

closed 2023-08-17 by mysterywolf