RT-Thread/rt-thread #7931: fatal error: dfs_posix.h: No such file or directory

sunlightxhw, 2023-08-08:
编译最新下载的rt thread lts v4.1.x (windows env bsp qemu-vexpress-a9)遇到这个问题：（看了下,这个问题很多人遇到，好像一直存在呢，没有完全解决……）
> scons                                                                                                     
scons: Reading SConscript files ...                                                                         
Newlib version:4.1.0                                                                                        
scons: done reading SConscript files.                                                                       
scons: Building targets ...                                                                                 
scons: building associated VariantDir targets: build                                                        
CC build\packages\gui_engine-latest\example\gui_demo.o                                                      
In file included from packages\gui_engine-latest\include/rtgui/image.h:28,                                  
                 from packages\gui_engine-latest\example\gui_demo.c:8:                                      
packages\gui_engine-latest\include/rtgui/filerw.h:39:10: fatal error: dfs_posix.h: No such file or directory
   39 | #include <dfs_posix.h>                                                                              
      |          ^~~~~~~~~~~~~                                                                              
compilation terminated.                                                                                     
scons: *** [build\packages\gui_engine-latest\example\gui_demo.o] Error 1                                    
scons: building terminated because of errors.

mysterywolf, 2023-08-08:
1. gui_engine这个软件包已经不再维护了
2. <dfs_posix.h>这个头文件已经被抛弃了 如果想要兼容的话，在组件里选择兼容模式 https://github.com/RT-Thread/rt-thread/pull/5277

sunlightxhw, 2023-08-09:
感谢回答。只是安装官网教程跑程序。其实GUI不用也可以。建议官网教程可以与时俱进一些。

closed 2023-08-09 by sunlightxhw