RT-Thread/rt-thread #7932: RT-Thread 5.0.1->ifconfig: network interface device list error.

sunlightxhw, 2023-08-08:
RT-Thread 5.0.1 @ windows env bsp qemu-vexpress-a9: （menuconfig 已开启ifconfig 并更新）
ifconfig: network interface device list error.
[image]

BernardXiong, 2023-08-09:
没网卡吧

sunlightxhw, 2023-08-11:
有啊：
[image]
我用4.1.x 就可以正常使用看到qemu rt 内IP地址。但是现在4.1.x 的问题是用vs code 无法调试：
[image]
运行qemu-dbg-.bat , 终端闪一下，我看没有显示出RT Thread Operating System 这几个字。说明应该是没有正常进入RT系统（？）导致没有启动gdb-server. 导致vs code 连接超时。应该是4.1.x dbg 模式的启动有些问题（？因为正常qemu.bat 脚本可以正常运行，可以正常ifconfig）现在不清楚怎么调试弄好了。不知道您是否有好的建议。

sunlightxhw, 2024-01-08:
menuconfig 把driver 这里的emac 打开就好了。