RT-Thread/rt-thread #7950: 错误码设定和返回规范

mysterywolf, 2023-08-12:
## 1  RT-Thread 有两种错误码
  - 一种是RT-Thread自己的错误码，如 `RT_ERROR` 、 `RT_EINVAL`等
  - 另一种是POSIX标准错误码 如 `ERROR` 、`EINVAL`等
  
若代码、组件是针对POSIX兼容设计的，则使用POSIX标准错误码，否则使用RT-Thread错误码。

例如 AT组件不是POSIX兼容设计，因此使用RT-Thread错误码，DFSv2是POSIX兼容设计，因此采用POSIX错误码

## 2 错误码的正负

- 以函数返回值方式无论POSIX错误码还是RT-Thread错误码都返回**负值**错误码 （除了 EOK 和RT_EOK）
- 以errno = 或rt_set_errno方式注入错误码，使用**正值**错误码， 如 `rt_set_errno(EINVAL)` 或 `rt_set_errno(RT_EINVAL)`

BernardXiong, 2023-08-13:
>若代码、组件是针对POSIX兼容设计的，

这个时候RT的错误码会被定义成POSIX的标准错误码

mysterywolf, 2023-08-13:
当POSIX打开之后，rt-thread错误码兼容POSIX错误码

liminghui12, 2025-06-24:
你好，使 用5.2.0 标准版  启posix 接口  测试dfsv2 open函数的错误码返回为负值，linux 中的posix 返回为正值，请问一下rtt里边有没有什么适配兼容办法 （因为要将原来linux开发环境的项目适配为rtthread 的posix 接口，如果更改应用代码，改动地方过多）

penguin216, 2026-11-10:
标准版 5.3.0 dfsv1/v2仍然存在返回负值错误码的问题：
./components/dfs/dfs_v2/src/dfs_posix.c
int open(const char *file, int flags, ...)
{
...

    if (file == NULL)
    {
        rt_set_errno(-EBADF);
        return -1;
    }

....
    fd = fd_new();
    if (fd >= 0)
    {
        df = fd_get(fd);
    }
    else
    {
        rt_set_errno(-RT_ERROR);
        return RT_NULL;
    }